FPGA-Research/FABulous #15: fabulous-dev\fabric\verilog_output: unconnected clock warning message (possible bug)

ruck314, 2022-01-26:
I am going through the quick start:
https://fabulous.readthedocs.io/en/latest/Usage.html#building-fabric

When I import the verilog_output files into Vivado, I am getting a warning message about the DSP_tile.v clock not being connected
```
WARNING: [VRFC 10-5021] port 'UserCLK' is not connected on this instance [XXXXX/verilog_output/DSP_tile.v:99]
```

nguyendao-uom, 2022-02-02:
Hi @ruck314, have you fixed or still encountering the issue?

ruck314, 2022-02-02:
@nguyendao-uom I have not fixed it.  But I could fork (or branch if you add me to the project) then investigate and try to fix it for you

nguyendao-uom, 2022-02-02:
@ruck314 Please check, just added you to the project

ruck314, 2022-12-01:
Closing this ticket because obsolete issue after `FABulous refactor`

closed 2022-12-01 by ruck314