FPGA-Research/FABulous #47: Move synthesis files out of nextpnr fork

TaoBi22, 2022-07-29:
The files that Yosys uses to synthesise designs for FABulous can be found inside [our nextpnr fork](https://github.com/FPGA-Research-Manchester/nextpnr/tree/adf49ce1393ae40e2e61b96b6f518a54e2427b4b), in `nextpnr/fabulous/synth/`. This isn't ideal - if a user only wishes to use the VPR flow, they have no choice but to clone the nextpnr submodule, even though they only need a small proportion of the files. In my opinion, this placement also just doesn't quite _make sense_ organisationally, since the files are equally related to both flows. 

Fixing this issue would mainly be some pretty simple reorganisation, so I've tagged it with 'good first issue', since it seems like a nice way to learn your way around the project a little for anyone hoping to get involved. If that's you then comment below and I'm more than happy to help!

A solution would probably require two pull requests, one to the nextpnr fork to move the files away, and another to the main FABulous repo to include them here.

- [ ] Move files into FABulous repo (probably merge this before we remove them from nextpnr so they're always available in at least one place)
- [ ] Move files out of nextpnr fork 
- [ ] Update `nextpnr/fabulous/fab_arch/fabulous_flow.sh` so that it's looking in the right place for the synthesis scripts
- [ ] Update our Yosys docs so that they point users to the correct filepath

gatecat, 2022-08-02:
It'd be nice to have a long-term goal of getting the synthesis support into upstream Yosys too (eventually, I would like to do a viaduct nextpnr 'uarch' that could be in upstream nextpnr too).

Given the configurability of fabulous, I would imagine that the Yosys pass would look something like `synth_fabulous -config /path/to/some/kind/of/config...`

TaoBi22, 2022-08-02:
@gatecat that definitely sounds great! Definitely think that a refactor to a viaduct uarch would be a great plan (as I think I said to you when I first learnt about it). I suspect that around now is going to be the point we really start bumping into the limitations of the Python API - for example, there's some desire at the moment for macro support which (based on pretty surface-level research so please correct me if I'm wrong!) seems much more doable using viaduct.

Agreed on the Yosys pass front - some parts of the config e.g. the primitive library should be pretty trivial to incorporate since we're obviously already going to have models of all the primitives, but I definitely think techmapping would benefit from being a bit smarter (been on my todo list for a while but have not as of yet got it done!), so that's potentially something that could fit nicely in there. Would also be nice to provide a good interface to inject custom rules, but that's probably a matter for future discussion!

gatecat, 2022-08-02:
>  for example, there's some desire at the moment for macro support which (based on pretty surface-level research so please correct me if I'm wrong!) seems much more doable using viaduct.

Yeah, there's full support for the 'relative constraints' mechanism that nextpnr uses to implement macros like LUT-MUX cascades, carry chains, etc, internally, in the viaduct framework but not easily with the old Python API.

In general, this is done by defining x-, y- and z- ('z' being the index of a bel within the tile) placement offsets against a root cell for all child cells inside a cluster.

gatecat, 2022-11-17:
Now that https://github.com/YosysHQ/yosys/pull/3548 upstream Yosys synthesis should be possible.

anudeepdharavathu, 2023-03-21:
[image]

Please help me with this issue.