FPGA-Research/FABulous #53: can not Linking CXX executable nextpnr-fabulous

LvmhZhD, 2022-08-20:
hi, i am in quick start , nextpnr-fabulous step ,when i code`make -j$(nproc)`,some maistakes happened,
`[  2%] Linking CXX executable nextpnr-fabulous
/usr/bin/ld: /home/***oss-cad-suite/lib/libdl.so.2: undefined reference to `_dl_vsym@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_dlclose@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_current_sigrtmax_private@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libpthread.so.0: undefined reference to `_dl_make_stack_executable@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_thread_freeres@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_siglongjmp@GLIBC_PRIVATE'
/usr/bin/ld: /home/zhaodonglv/oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_pthread_init@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_allocate_rtsig_private@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libdl.so.2: undefined reference to `_dl_addr@GLIBC_PRIVATE'
/usr/bin/ld: /home/***oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_dlopen_mode@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_dlsym@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_longjmp@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libdl.so.2: undefined reference to `_dl_sym@GLIBC_PRIVATE'
/usr/bin/ld: /home/***/oss-cad-suite/lib/libpthread.so.0: undefined reference to `__libc_current_sigrtmin_private@GLIBC_PRIVATE'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/nextpnr-fabulous.dir/build.make:651: nextpnr-fabulous] Error 1
make[1]: *** [CMakeFiles/Makefile2:101: CMakeFiles/nextpnr-fabulous.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
`
how can i fix it?

gatecat, 2022-08-20:
It looks like you've been running `make` with oss-cad-suite in your path. Try running it again on a clean build without oss-cad-suite in your path.

LvmhZhD, 2022-08-20:
> It looks like you've been running `make` with oss-cad-suite in your path. Try running it again on a clean build without oss-cad-suite in your path.

Wow,i fix it by you methods, After installing Yosys and litex, i did not deactivate OSS-CAD suite without any hints， thank you !!

gatecat, 2022-08-20:
No problem! As an aside to the issue here, hopefully, in the not so distant future, https://github.com/YosysHQ/nextpnr/pull/1015 will be merged and it will be possible then to get nextpnr with fabulous support prebuilt in the oss-cad-suite, too.

closed 2022-08-21 by TaoBi22