FPGA-Research/FABulous #57: C

LvmhZhD, 2022-10-13:


gatecat, 2022-10-13:
Hi @LvmhZhD!

It is probably the case that the primitive view that Yosys is given in the prims.v file used for synthesis doesn't have the pass, because we've never really supported the output register feature in synthesis and PnR yet.

What exactly are you trying to compile in `user_design/*.v`? A mention of `W_IO` makes it seem like you are trying to use the synthesis script for building bitstreams to run on the fabric to instead build the fabric itself, which isn't going to fly (the fabric verilog is for a VLSI flow like OpenLane.)

LvmhZhD, 2022-10-13:
> 

Ok. I get. I compiled all the Verilog files of  eFPGA generated from FABulous to verify its function on the board. I saw A introduction about FABulous shows it is supported to verify it on the board. 
 But It seems i confused verification’s step on the board.

LvmhZhD, 2022-10-15:
> 

Sorry , i am confused still. And i  have two questions.
1. We use the FABulous to generate the verilog files after`run_fab_flow.sh`,is the generated files called the fabric verilog?
2. To verify the eFPGA function, we plan to implement it in our Xliinx FPGA board(maybe teh ZCU 104...). We want a bitstream. In quick start,`run_fab_flow.sh <rows> <cols>` does not generate Nextpnr and vpr model files as expected in `generating bitstream`. So we follow `RTL to Bitstream` steps,it looks the same as `generating bitstream `in `Quick Start`.However,we fail to Synthesis the verilog file, the encounter problem is what i emntioned before `ERROR: Module IO_1_bidirectional_frame_config_pass' referenced in module W_IO' in cell 'Inst_B_IO_1_bidirectional_frame_config_pass' does not have a port named 'UserCLK'.`the verilog files in `/user_design/*.v` are all the verilog output(including `eFPGA_top.v`,`Frame_select....V`,`N_term....v`,etc.
3. 
And I am puzzled about `because we've never really supported the output register feature in synthesis and PnR yet.`, is that means we cannot generate bitstream with prims.v or what `verilog -lib `we need in `synth_fabulous_dffesr.tcl`. Do you have any idea?

gatecat, 2022-10-17:
1. Yes
2. This seems like a problem, but you can maybe look at how the Icarus smoketest runs synthesis and PnR: https://github.com/FPGA-Research-Manchester/FABulous/commit/b710369fe939f7c1d4ca4498a254165f78c6e7f2 and https://github.com/FPGA-Research-Manchester/FABulous/tree/master/fabric_files/fabric_icarus_example - if this isn't working then you've hit a bug.
3. No, generating bitstreams for the fabric should work fine - it's just that one specific feature of the clock pin that's missing. The problem is, you are trying to compile the whole eFPGA through this flow that's targetting the eFPGA, rather than some test design like a counter (this script is for building designs to run on the eFPGA, _not_ for building a bitstream to test the eFPGA on another fabric). 

Although there is a planned support for emulating the eFPGA on another FPGA fabric in the future, this isn't really supported at the moment and testing it in simulation instead (e.g. Icarus Verilog as in our CI, Vivado Simulator, ModelSim etc) is the recommended way to go.

Do let me know if I can help more with your planned use of FABulous though - my email is in my github bio if you want to set up a call or something.

LvmhZhD, 2022-10-17:
Okay！Many thanks

closed 2025-01-27 by KelvinChung2000