FPGA-Research/FABulous #67: `simulation smoketest` bugs

ruck314, 2022-12-05:
I was running into issues when I was simulating my eFPGA yesterday.  I went back to the original icarus simulation and updating to highlight the issues that I am seeing:
https://github.com/FPGA-Research-Manchester/FABulous/commit/1496c92a36bbc47b35dc62691dbde1ee2604aaec

I was able to reproduce the issues in icarus simulations.  If I set the upper 10-bits of `io_out` to a non-zero value, I am still getting zeros for those bits.  If I compare the io_oeb DUT to eFPGA outputs, then are totally off, which is highlighted in the new simulation print statement. 

```
+ python3 makehex.py test_design/counter.bin 16384 bitstream.hex
+ vvp fab_tb.vvp
fabric(I_top) = 0x0000000 gold = 0x48c0000, fabric(T_top) = 0xfffffff gold = 0x0000001
fabric(I_top) = 0x0000004 gold = 0x48c0004, fabric(T_top) = 0xfffffff gold = 0x0000001
fabric(I_top) = 0x0000008 gold = 0x48c0008, fabric(T_top) = 0xfffffff gold = 0x0000001
...
...
...
fabric(I_top) = 0x0000184 gold = 0x48c0184, fabric(T_top) = 0xffffe7f gold = 0x0000001
fabric(I_top) = 0x0000188 gold = 0x48c0188, fabric(T_top) = 0xffffe7f gold = 0x0000001
fabric(I_top) = 0x000018c gold = 0x48c018c, fabric(T_top) = 0xffffe7f gold = 0x0000001
FATAL: fabulous_tb.v:78:
       Time: 206950000 Scope: fab_tb
```

gatecat, 2022-12-05:
Thanks for reporting, I will investigate this and get back to you.

gatecat, 2022-12-06:
Should hopefully be fixed by https://github.com/YosysHQ/nextpnr/pull/1056, can you have a try?

gatecat, 2022-12-06:
OK I see one remaining issue that tristates are inverted inside the fabric - I will add a fix for that too

ruck314, 2022-12-06:
@gatecat I merged my dev branch with origin/master.  Next, I created a "token" pull request to kick off the Github Action CI for my dev branch (help with communicating the issue).  Unfortunately, I am still observing the issue:
https://github.com/FPGA-Research-Manchester/FABulous/actions/runs/3629428860/jobs/6121620824

gatecat, 2022-12-06:
As the nextpnr build used in the CI is only updated nightly, unfortunately you'll need to wait until tomorrow morning (Europe time) for the fix to be applied.

ruck314, 2022-12-07:
@gatecat I have confirmed your https://github.com/YosysHQ/nextpnr/pull/1056 fixes the issue
https://github.com/FPGA-Research-Manchester/FABulous/actions/runs/3630033877/jobs/6145881432
I recommend that we marge PR #70.  I will close the issue ticket since this is resolved now.

closed 2022-12-07 by ruck314