FPGA-Research/FABulous #79: VPR is not working with Yosys 0.19+0

KelvinChung2000, 2023-01-13:
When running the VPR flow using Yosys 0.19+0 yields the following error. 
```
Error 1: 
Type: Blif file
File: ../nextpnr/fabulous/fab_arch/sequential_16bit_en.blif
Line: 105
Message: Failed to find matching architecture model for 'LUTFF_ESR'
```
Looks like Yosys mapping has some problem causing this error.

gatecat, 2023-01-13:
`LUTFF_ESR` mapping shouldn't be enabled in the normal synthesis script, which one are you running?

KelvinChung2000, 2023-01-13:
Using `./nextpnr/fabulous/fab_arch/fabulous_flow.sh`, the `.json` changed to `.blif`. 
The yosys command is the following:
`yosys -p "tcl ../synth/synth_fabulous_dffesr.tcl 4 ${1} ${1}.blif" ${1}.v -l ${1}_yosys_log.txt`

TaoBi22, 2023-01-13:
Since we now have an upstream Yosys synthesis pass, is it worth just removing the old synthesis script and the surrounding documentation, and recommending the upstream Yosys pass instead? (As a note, based on a quick test just now I'm also getting issues with the upstream pass and VPR but I'll make another issue for that once I've explored it more).

TaoBi22, 2023-01-13:
Never mind, the upstream pass actually seems to work fine and the issues were on my end!

TaoBi22, 2023-01-17:
Should now be fixed in #80

closed 2023-01-17 by TaoBi22