FPGA-Research/FABulous #82: Installation Issues

RobB720, 2023-02-07:
Hi,

I'm trying to install FABulous on RHEL8. I had install various packages etc. and made progress but have hit an issue at the "make -j$(nproc)" command:

[  7%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/archcheck.cc.o
[ 13%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/bits.cc.o
[ 21%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/design_utils.cc.o
[ 10%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[ 21%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/command.cc.o
[ 10%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/basectx.cc.o
[ 21%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/embed.cc.o
[ 21%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/context.cc.o
[ 23%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/handle_error.cc.o
[ 26%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/idstring.cc.o
[ 28%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/idstringlist.cc.o
[ 31%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/log.cc.o
[ 34%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/nextpnr.cc.o
[ 36%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/nextpnr_assertions.cc.o
[ 39%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/nextpnr_namespaces.cc.o
[ 42%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/nextpnr_types.cc.o
[ 44%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/place_common.cc.o
[ 47%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/placer1.cc.o
[ 50%] Linking CXX executable bbasm
/usr/bin/ld: CMakeFiles/bbasm.dir/main.cc.o: undefined reference to symbol '_ZNKSt3_V214error_category10_M_messageB5cxx11Ei@@GLIBCXX_3.4.21'
//usr/lib64/libstdc++.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [bba/CMakeFiles/bbasm.dir/build.make:100: bba/bbasm] Error 1
make[1]: *** [CMakeFiles/Makefile2:153: bba/CMakeFiles/bbasm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 52%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/placer_heap.cc.o
[ 55%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/property.cc.o
[ 57%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/pybindings.cc.o
[ 60%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/router1.cc.o
[ 63%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/router2.cc.o
[ 65%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/sdf.cc.o
[ 68%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/str_ring_buffer.cc.o
[ 71%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/svg.cc.o
[ 73%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/timing.cc.o
[ 76%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/common/timing_opt.cc.o
[ 78%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/3rdparty/json11/json11.cpp.o
[ 81%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/json/jsonwrite.cc.o
[ 84%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/frontend/json_frontend.cc.o
[ 86%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/fabulous/arch.cc.o
[ 89%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/fabulous/arch_pybindings.cc.o
[ 92%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/fabulous/cells.cc.o
[ 94%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/fabulous/main.cc.o
[ 97%] Building CXX object CMakeFiles/nextpnr-fabulous.dir/fabulous/pack.cc.o
/home/rbeat/FABulous/nextpnr/fabulous/pack.cc:308:13: warning: ‘void nextpnr_fabulous::promote_globals(nextpnr_fabulous::Context*)’ defined but not used [-Wunused-function]
 static void promote_globals(Context *ctx)
             ^~~~~~~~~~~~~~~
[100%] Linking CXX executable nextpnr-fabulous
/usr/bin/ld: /tmp/ccMMGu8x.ltrans30.ltrans.o: undefined reference to symbol '_ZNKSt3_V214error_category10_M_messageB5cxx11Ei@@GLIBCXX_3.4.21'
//usr/lib64/libstdc++.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/nextpnr-fabulous.dir/build.make:651: nextpnr-fabulous] Error 1
make[1]: *** [CMakeFiles/Makefile2:127: CMakeFiles/nextpnr-fabulous.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

Looking in the CMakeError.log file:

Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/rbeat/FABulous/nextpnr/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_da4b6/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_da4b6.dir/build.make CMakeFiles/cmTC_da4b6.dir/build
gmake[1]: Entering directory '/home/rbeat/FABulous/nextpnr/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_da4b6.dir/src.c.o
/usr/bin/cc -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_da4b6.dir/src.c.o -c /home/rbeat/FABulous/nextpnr/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_da4b6
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_da4b6.dir/link.txt --verbose=1
/usr/bin/cc CMakeFiles/cmTC_da4b6.dir/src.c.o -o cmTC_da4b6 
CMakeFiles/cmTC_da4b6.dir/src.c.o: In function `main':
src.c:(.text+0x2d): undefined reference to `pthread_create'
src.c:(.text+0x39): undefined reference to `pthread_detach'
src.c:(.text+0x45): undefined reference to `pthread_cancel'
src.c:(.text+0x56): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
gmake[1]: *** [CMakeFiles/cmTC_da4b6.dir/build.make:99: cmTC_da4b6] Error 1
gmake[1]: Leaving directory '/home/rbeat/FABulous/nextpnr/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:127: cmTC_da4b6/fast] Error 2

Source file was:
#include <pthread.h>

static void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/rbeat/FABulous/nextpnr/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_275bf/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_275bf.dir/build.make CMakeFiles/cmTC_275bf.dir/build
gmake[1]: Entering directory '/home/rbeat/FABulous/nextpnr/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_275bf.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create -o CMakeFiles/cmTC_275bf.dir/CheckFunctionExists.c.o -c /usr/share/cmake/Modules/CheckFunctionExists.c
Linking C executable cmTC_275bf
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_275bf.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create CMakeFiles/cmTC_275bf.dir/CheckFunctionExists.c.o -o cmTC_275bf  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
gmake[1]: *** [CMakeFiles/cmTC_275bf.dir/build.make:99: cmTC_275bf] Error 1
gmake[1]: Leaving directory '/home/rbeat/FABulous/nextpnr/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:127: cmTC_275bf/fast] Error 2

From the above file it seems to be an issue with "pthread"

I am certainly not a software/linxu expert(!) but the following stack overflow post has a similar error that is overcome with a compiler command line option which might help:

https://stackoverflow.com/questions/1662909/undefined-reference-to-pthread-create-in-linux

Any ideas how I can fix this?

Best regards,
Rob.

gatecat, 2023-02-07:
The FABulous nextpnr fork is no longer used, please try with upstream nextpnr or, if it works for you, prebuilt binaries from [oss-cad-suite](https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2023-02-07)

Do let me know if you're still hitting problems with upstream nextpnr.

RobB720, 2023-02-07:
Thanks for the reply, but can you explain what you mean by "upstream nextpnr"? (Apologies, I don't have much experience of open-source tools...)

gatecat, 2023-02-07:
The main, publicly released version of nextpnr: https://github.com/YosysHQ/nextpnr

(or the prebuilt binaries I mentioned)

closed 2023-11-10 by TaoBi22