FPGA-Research/FABulous #112: Demo W_IO problem/questions

RobB720, 2023-09-20:
Hi,

I'm using the demo architecture provided as an instructional basis for my own architecture along with the online documentation.

I am a bit confused about the files provided for the demo/W_IO tile. The fabric.csv points at the folllowing files with BEL commands:

./Tile/W_IO/IO_1_bidirectional_frame_config_pass.v
./Tile/W_IO/Config_access.v
./Tile/W_IO/W_IO_switch_matrix.list

In the "demo" directory the file ./Tile/W_IO/Config_access.v does not exist but ./Tile/W_IO/Config_access.v_safe is there - is this file correct?

The file IO_1_bidirectional_frame_config_pass.v does not seem to use configuration bits, so why is a switch matrix required?

Looking to set my own IO tile up so any explanatory help gratefully received!

Regards,
Rob.

gatecat, 2023-09-20:
The switch matrix is used to select which signals are used to drive the IO output/tristate, and where the IO input is routed to.

The file in git is named [Config_access.v](https://github.com/FPGA-Research-Manchester/FABulous/blob/master/fabric_files/FABulous_project_template_verilog/Tile/W_IO/Config_access.v), I'm not sure what is adding the _safe postfix, could it be an antivirus tool or some other script that's running?

RobB720, 2023-09-21:
Thanks for the reply.

In my copy of IO_1_bidirectional_frame_config_pass.v the parameter NoConfigBits = 0 (in fact it's commented out) - hence my confusion.

Perhaps my demo area is old/corrupted - however I can't find it in the Github repository (sorry, I have very little experience using Github!). Can you point me at it?

Also, are there any other example implementations using FABulous? If so where can I find those?

Many thanks,
Rob.

closed 2025-01-27 by KelvinChung2000