FPGA-Research/FABulous #124: Clarification of and differences between ".fasm" and "_des.fasm" files

IAmMarcelJung, 2023-11-06:
In the flow, some commands use a ".fasm" file and some use a "_des.fasm" file. Some commands, like "gen_bitstream_binary" expect the specification of a ".fasm" file, but use "_des.fasm" internally. When comparing the CLI commands to the manual commands, the CLI way with "place_and_route_npnr"  creates a "_des.fasm" (although it expects just a ".fasm" file in the command), while the manual way  creates just a ".fasm" file (although this could obviously be changed by the user of course, [see documentation](https://fabulous.readthedocs.io/en/latest/FPGA-to-bitstream/Nextpnr%20compilation.html)).

The command gen_bitStream_binary also creates a ".fasm" file, but the content does not quite look like a ".fasm" file as specified in the  [FASM documentation](https://fasm.readthedocs.io/en/latest/index.html), at least for my understanding. The file consists of 14 blocks that look like this (all the same except for the first line, where just the 14 changes down to 1 until the last block):
```
X9Y14,RAM_IO,9,14
frame0,0,32,00000000000000000000000000000000
frame1,1,32,00000000000000000000000000000000
frame2,2,32,00000000000000000000000000000000
frame3,3,32,00000000000000000000000000000000
frame4,4,32,00000000000000000000000000000000
frame5,5,32,00000000000000000000000000000000
frame6,6,32,00000000000000000000000000000000
frame7,7,32,00000000000000000000000000000000
frame8,8,32,00000000000000000000000000000000
frame9,9,32,00000000000000000000000000000000
frame10,10,32,00000000000000000000000000000000
frame11,11,32,00000000000000000000000000000000
frame12,12,32,00000000000000000000000000000000
frame13,13,32,00000000000000000000000000000000
frame14,14,32,00000000000000000000000000000000
frame15,15,32,00000000000000000000000000000000
frame16,16,32,00000000000000000000000000000000
frame17,17,32,00000000000000000000000000000000
frame18,18,32,00000000000000000000000000000000
frame19,19,32,00000000000000000000000000000000
```

Could someone explain the difference between a "_des.fasm" and ".fasm" and when which one is used? Additionally, is there a reason why sometimes a ".fasm" file has to be specified, but internally a "_des.fasm" file is used and its existence is expected? Is the distinction even necessary?

Thanks in advance for any help!

IAmMarcelJung, 2023-11-09:
I understood a few things by reading the documentation. The structure of the file looks a lot like the file shown for the bitstream remapping (or any of the <tile_descriptor>_ConfigMem.csv files), although not exactly the same, since the configuration bit range is missing and the "headline" for each block does not look the same. 

I still don't understand why this file is named .fasm. In the documentation it says that it should be a .csv file which it also is. Is it intentional that it is named .fasm instead of .csv or just a bug/mistake? To me it seems to be intentional, since the _des.fasm file could then just be named .fasm, but maybe someone knows more about this.

KelvinChung2000, 2023-11-09:
The `_des` is just for classifying user design. You should see that the use `bit_gen.py` is inconsistent with the rest of the FABuslous CLI. This is because I did not write that script, and I do not understand how that script produce the bitstream, so I shouldn't touch anything related. 

I likely put that suffix in to avoid that "external" script overwriting the user design FASM file. 

If you need to call nextpnr outside the CLI, you can emit the file without the `_des` suffix and call `bit_gen.py` using whatever you named the FASM file.

IAmMarcelJung, 2023-11-10:
Thanks for the replies and also the fix!
> The `_des` is just for classifying user design. You should see that the use `bit_gen.py` is inconsistent with the rest of the FABuslous CLI. This is because I did not write that script, and I do not understand how that script produce the bitstream, so I shouldn't touch anything related.

I don't quite understand what you mean by saying that ```bit_gen.py``` is inconsistent with the CLI. The CLI calls ```bit_gen.py``` in ```do_gen_bitStream_binary```, so essentially the manual method and the CLI method are the same, if you specify a ```.fasm``` file with the ```_des``` suffix for the  manual method:

```
        runCmd = ["python3", f"{fabulousRoot}/fabric_cad/bit_gen.py",
                  "-genBitstream",
                  f"{self.projectDir}/{path}/{name}_des.fasm",
                  f"{self.projectDir}/.FABulous/bitStreamSpec.bin",
                  f"{self.projectDir}/{path}/{name}.bin"]
```

> If you need to call nextpnr outside the CLI, you can emit the file without the _des suffix and call bit_gen.py using whatever you named the FASM file.

So the ```_des.fasm``` file refers to the file generated by FABulous and any ```.fasm``` file without the ```_des``` suffix refers to a file created/edited by the user?

KelvinChung2000, 2023-11-10:
You can see that the `bit_gen.py` is used by calling a subprocess that is Python code rather than calling an internal function. This is what I mean by inconsistent. 

If you want to distinguish files generated by FABulous and files created/edited by the user, you can use this as the indicator. We could name all FABulous internally generated files if needed, starting with something like `FAB_`. But what is the actual use case for such a change?

IAmMarcelJung, 2023-11-10:
>You can see that the ```bit_gen.py```  is used by calling a subprocess that is Python code rather than calling an internal function. This is what I mean by inconsistent.

Yeah makes sense!

>If you want to distinguish files generated by FABulous and files created/edited by the user, you can use this as the indicator. We could name all FABulous internally generated files if needed, starting with something like FAB_. But what is the actual use case for such a change?

Okay I think I got the difference now, thanks for the explanation! I don't see a use case giving FABulous generated files a special suffix.  I think my initial misunderstanding was caused by the bug fixed in #128 where both a ```.fasm``` and a ```_des.fasm``` file were generated. Maybe we should still document that FABulous creates a ```_des.fasm``` file and a user should name the self-created/edited file just ```.fasm```?

Is it intended that you can only use your own file with the manual method and not the CLI? As mentioned in my initial post ```gen_bitStream_binary``` appends the module specified with ```_des``` and therefore looks for a ```_des.fasm``` file. If it is intended everything is fine, if not we should change it so the user can either specify either his custom ```.fasm``` or the generated ```_des.fasm``` file.

KelvinChung2000, 2023-11-10:
The philosophy for the CLI is if you don't know what you are doing, then work within the CLI. 

We try to ensure things will work if any work is done using the CLI exclusively. Ideally, the CLI should work like a complete CAD tool like Vivado, and we produced enough options and toggles so that you can do everything in the CLI. But getting to that level of completeness is challenging and requires much work. The manual flow is the way to go for people who need to do more than what the CLI provide.

TaoBi22, 2023-11-10:
> Is it intended that you can only use your own file with the manual method and not the CLI? As mentioned in my initial post `gen_bitStream_binary` appends the module specified with `_des` and [...]

IMO it's pretty clear-cut we should just ditch the `_des` suffix. It was just there to avoid a conflict that we no longer have to worry about, and the fact that it's appended to the end of the name given to the tool is inevitably going to confuse more people as it's very counterintuitive. @KelvinChung2000 if you agree I'll get rid of it?

KelvinChung2000, 2023-11-10:
True, getting rid of it will be easier.

IAmMarcelJung, 2023-11-10:
I also think its better with ```_des``` removed. Thanks for the clarification!

closed 2023-11-10 by IAmMarcelJung