FPGA-Research/FABulous #134: Non exisiting directory in documentation of the simulation

IAmMarcelJung, 2023-11-17:
In the documentation of the simulation setup, there is a path ```$FAB_ROOT/fabric_files/fabric_icarus_example/``` given, but the directory ```fabric_icarus_example``` does not exists for me. In fabric_files I only have the directories ```FABulous_project_template_``` for both verilog and VHDL and a ```generic``` directory.  Is the directory intended to exist or is the documentation of the simulation setup outdated?

However I have a ```Test``` directory in the project directory which seems to include all the test scripts, a test bench and a ```README.MD``` which to me indicates that the documentation is outdated. 

Maybe @TaoBi22  can help, since you mentioned that you worked on the documentation before?

KelvinChung2000, 2023-11-21:
We have updated the testing structure to follow the "project" structure. The documentation will need to be updated.

TaoBi22, 2023-11-23:
Hey @IAmMarcelJung, sorry for the delayed response! This is indeed a holdover from the old structure of the project, I'll get it fixed. Thanks a lot for catching it!

closed 2023-11-23 by TaoBi22

IAmMarcelJung, 2023-11-23:
Nice, thanks for fixing!