FPGA-Research/FABulous #139: What do NN, EE, SS and WW stand for?

IAmMarcelJung, 2023-11-29:
In the [documentation](https://fabulous.readthedocs.io/en/latest/fabric_definition.html#wires), the naming scheme for ```source_name``` and ```destination_name``` of a wire is explained.  According to the definition, for a wire named e.g. ```NxBeg```, ```x``` should reflect how far the wire is routed, with ```x ∈ {1,2,4,6}```. However, in ```fabric.csv```, ```x``` can also be ```N```, ```E```, ```S``` or ```W```, depending on the direction, so a wire can have e.g. the name ```NNBeg```.  I could not find a definition for this type of wire in the definition.

Can anybody tell me what it stands for or point me to where it is explained in the documentation, if it is explained there?

Thanks in advance!

IAmMarcelJung, 2023-11-30:
I have to somewhat correct myself. It seems it has nothing to do with the value of ```x```, since ```NNBEG``` is not a valid name, but ```NN4BEG``` would be. The letters ```N```, ```E```, ```S```, ```W``` are therefore an addition, not a replacement. The addition is also sometimes made to ```JUMP``` wires, so ```JN```, ```JE```,  ```JS```, ```JW```, exist,  but ```JJ``` does not exists.

Sorry for the confusion! My main question from the title still remains.

KelvinChung2000, 2023-12-01:
I have no deep knowledge of the FABulous architecture. I recalled that the NN, EE, indicates a double wire. You will not have things like JN or JE because the jump wire is internal to the tile. @gatecat will be able to give a better answer.

IAmMarcelJung, 2023-12-01:
Thanks for your reply! I also thought that JN or JE don't exist, but they exist in fabric.csv and also appear in the ```.fasm``` file. That's one part that confused me.

KelvinChung2000, 2023-12-01:
Maybe look into the `.list` file. That might be useful.

IAmMarcelJung, 2024-03-08:
I just talked to @dirk-koch and he cleared things up. The naming is historically grown (as @everythingelsewasalreadytaken also had previously guessed in a private talk). One naming scheme was used when only smaller fabrics were built. The other naming scheme came on top when bigger fabrics had to be built and more routing needed to be added. To distinguish between "old" and "new" routing resources, a different naming scheme was used rather than using the same one. So in short, those wires with two letters do not differ in functionality from those with one letter.

closed 2024-03-08 by IAmMarcelJung