FPGA-Research/FABulous #142: "eFPGA_top" parameters for larger fabric

anudeepdharavathu, 2023-12-10:
eFPGA_top #(
        .include_eFPGA(1),
        .NumberOfRows(ROW_WIDTH),
        .NumberOfCols(COL_WIDTH),
        .FrameBitsPerRow(32),
        .MaxFramesPerCol(20),
        .desync_flag(20),   
        .FrameSelectWidth(5),
        .RowSelectWidth(5)
    )     
    
    which of these parameters need to be changed for larger fabrics say 256 IO tiles?
    the default parameters do not seem to work.
    I changed ROW_WIDTH but what about FrameSelectWidth and RowSelectWidth?
    I don't see any documentation on them.

KelvinChung2000, 2023-12-12:
The parameterisation of the eFPGA is not from the Verilog code itself but from the CSV file.

anudeepdharavathu, 2023-12-12:
FrameSelectWidth and RowSelectWidth parameters are not available in the .csv file.

when I generate large fabric, it doesn't work. Reducing the size works fine so I am requesting the procedure to optimize parameters for larger fabric.
I have only increased the row size to accommodate more IO tiles.
Do I need to change anything else other than ROW_WIDTH, BITSTREAM length, I/O register width in the test bench, top_wrapper.v file ?

KelvinChung2000, 2023-12-14:
I have looked into the generator. The `FrameSelectWidth` and `RowSelectWidth` are not parsed. This will require a fix in the CSV parser. For now, you will need to change the `FrameSelectWidth` and `RowSelectWidth` manually in the Verilog.

anudeepdharavathu, 2023-12-15:
can you please explain how would these values change with increase in fabric size?

EverythingElseWasAlreadyTaken, 2025-06-11:
This is also discussed in #168. 
I'll close this issue for now, please reopen if there are any updates on this.