FPGA-Research/FABulous #148: Exception messages not printed

IAmMarcelJung, 2024-02-02:
Since #130, the exception message are not printed in the shell anymore. Since we already have logging, the exception message might also not be need and could therefore be removed. 

If we however need the messages, something like 
```
except Exception as e:
    print(e)
    return False
```
inside ```onecmd``` would again print the exception message. We should therefore decide if either omitting or printing the exception message is preferred.

IMHO it would be better to print the messages, but any other suggestions (also different ones to these two options) are welcome as always.

KelvinChung2000, 2024-02-06:
Printing the exception message will be better, as sometimes the cause of the error might be absent. Directly logging the error message will also be helpful.

KelvinChung2000, 2024-02-16:
fixed on #152

closed 2024-02-16 by KelvinChung2000