FPGA-Research/FABulous #151: FABulous.py + multiple verilog files in the user_design

ruck314, 2024-02-08:
I have multiple verilog files that I want to synthesis into a bitstream for my eFPGA:
```bash
$ ll user_design/
total 94
-rw-rw-rw- 1 ruckman re 13571 Feb  7 16:38 my_prj_decision_function_1.v
-rw-rw-rw- 1 ruckman re 10370 Feb  7 16:38 my_prj_decision_function.v
-rw-rw-rw- 1 ruckman re  1580 Feb  7 16:38 my_prj_mux_63_22_1_1.v
-rw-rw-rw- 1 ruckman re  2148 Feb  7 16:38 my_prj_mux_94_22_1_1.v
-rw-rw-rw- 1 ruckman re  7553 Feb  7 16:38 my_prj_regslice_both.v
-rw-rw-rw- 1 ruckman re 39120 Feb  7 16:38 my_prj.v
-rw-r--r-- 1 ruckman re  1471 Feb  7 17:32 top.v
-rw-r--r-- 1 ruckman re 15117 Feb  7 17:13 top_wrapper.v
```
When I do `run_FABulous_bitstream {npnr user_design/top.v}` using a TCL script with FABulous.py, I get an error that `ERROR: Module `\my_prj' referenced in module `\top' in cell `\my_proj_i' is not part of the design.`.  This makes sense because I have not define the other verilog files in the project.

How can I use `FABulous.py` to add all these other `my_prj*.v" files to do the FABulous bitstream generation?

KelvinChung2000, 2024-02-08:
The TCL script runs under the FABulous shell, and the shell command only supports one file. If you want to do multiple files, you must do it manually. 

Extending the `do_synthesis` command will allow you to include multiple files. If "design" is a directory, it will find all the files ending with `.v` and include them in the synthesis command.

ruck314, 2024-02-12:
@KelvinChung2000 Are there any plans for supporting VHDL only or mixed Verilog/VHDL designs? I am just trying to get a sense of what is supported and not supported in the synthesis process.

closed 2024-02-12 by ruck314

KelvinChung2000, 2024-02-12:
The main limiting factor is Yosys. As long as there are ways to let Yosys read the RTL, Yosys should be able to synthesise the design. If you want it to be supported with FABulous, will need to get someone to update the installing doc and update the FABulous synthesis command. I am only passively maintaining the project, not actively developing anything.