FPGA-Research/FABulous #157: Orientations of basic tiles

mole99, 2024-02-19:
Upload all orientations for basic tiles, such as `[N|E|S|W]_IO`.

This makes it easier for new users to modify the fabric and get started using FABulous.

AhmadHouraniah, 2024-10-30:
Hello,

I'm working on modifying my architecture to support IOs on all sides, but I’m uncertain about the specific adjustments required for the N/S IOs. From what I understand, I need to create separate folders for each IO type (N, E, S, W) and define the associated switch matrices, then add these to the fabric.csv file.

Could you confirm if this is the correct approach, or if there’s a more efficient way to configure IOs on all sides? Manually defining each switch matrix feels error-prone, so any guidance would be appreciated. Additionally, if you have example switch matrices or a fabric.csv file that includes these configurations, that would be incredibly helpful.

Thanks!

KelvinChung2000, 2024-10-30:
With the current implementation, you will need to create an individual IO tile for each side. This is because the wire going in and out of the tile will be oriented differently on each side. 

I have no clean solution for describing IO tiles since each fabric switch matrix is unique and IO tile will be unique.   

This is one example of the IO tile. 
https://github.com/gatecat/fabulous_mpd/tree/work/fabric/Tile/S_term_single

mole99, 2024-10-30:
Hi @AhmadHouraniah, I did this a while ago for the missing IO tiles: [io_tiles.zip](https://github.com/user-attachments/files/17574821/io_tiles.zip)

I would appreciate if you could have a look at the config files before using them and let me know if you find any error :)

AhmadHouraniah, 2024-10-30:
Thank you for your help and for sharing the files. I will review the configurations and let you know if I find any issues.