FPGA-Research/FABulous #158: Code Formatter

mole99, 2024-02-19:
Use a code formatter to format the codebase.

Before any discussions come up about which coding style to use, I would suggest just to use [Black](https://github.com/psf/black):

> Black is the uncompromising Python code formatter. By using it, you agree to cede control over minutiae of hand-formatting. In return, Black gives you speed, determinism, and freedom from pycodestyle nagging about formatting. You will save time and mental energy for more important matters.

KelvinChung2000, 2024-02-19:
Formatting and style guides are things that need work on. I am using a heavily modified autopep8. 

Is there an easy way to set up the style? i.e. after cloning, can the code automatically apply to the style guide with an editor like vscode?

mole99, 2024-02-20:
Well, you can just install black via pip: `pip install black` and then format files like so: `black my_file.py`.

I don't use vscode, but there seems to exist an extension for it: [Black Formatter](https://marketplace.visualstudio.com/items?itemName=ms-python.black-formatter). Support for other editors does exist too.

I can take a look at integrating black with the FABulous codebase. Basically the CI runs black and fails if the code has not been formatted. If desired, reviewdog can create comments on the PR with the necessary changes. Or the developer can just fix it themselves by running black.

IAmMarcelJung, 2024-07-30:
Hi, this could be closed since we use black in the GitHub actions, right? :)

IAmMarcelJung, 2024-07-30:
My bad, forgot that this is still only in the dev branch. Sorry for the disturbance.

mole99, 2024-07-30:
No worries! Any news on when dev will be merged?

IAmMarcelJung, 2024-07-30:
Unfortunately I don't know anything about when the merge is targeted :/

KelvinChung2000, 2024-07-31:
I am trying to push a stable release with what we have now. I really would like this done asap, ideally weeks after the summer school. Otherwise, another workshop might come up, and we once again need to delay this

closed 2025-01-27 by KelvinChung2000