FPGA-Research/FABulous #160: Missing configuration for CPU_IO

mole99, 2024-02-21:
The `CPU_IO` tile is currently not available for Verilog and misses configurations in VHDL.

Instead of removing this tile, it would be nice if its functionality could be restored.

dirk-koch, 2024-03-06:
While our I/O stuff works, I am not super happy with how it is used.
Meaning, its not good to provide different I/O BELs whenever something minor changes.
I want this more like something that you can specify an I/O BEL in the form #in, #out, #inout, #config_bits with a name prefix and fundamentally that's it.
Of course if you have other requirements, you will still be able to provide BEL I/Os as RTL code...
There is somebody working on it right now.

mole99, 2024-03-07:
Hello Dirk,
That sounds like a good solution. Great to hear that somebody is already working on it. Can't wait to try it out 👍️

EverythingElseWasAlreadyTaken, 2025-05-28:
If I recall right, the CPU_IO is similar to our RAM_IO.

I'll close this issue, please reopen if I've missed anything here.

closed 2025-05-28 by EverythingElseWasAlreadyTaken