FPGA-Research/FABulous #162: Question: Clock distribution

mole99, 2024-02-22:
Each tile has a clock input `UserCLK` and a clock output `UserCLKo` to distribute the clock to the next tile. Thus, this leads to a clock distribution as seen on the left image:

[image]

This is not a problem if the design gets flattened. Then the clock tree distributes the clock to all tiles equally.

Normally, when implemented in an ASIC, the tiles are hardened as individual macros and later stitched together. As I understand it, this would mean that the clock is delayed between each rows of macros. Basically, each row has its own clock domain ;)
To fix this, the clock must be distributed in a star-shaped pattern to each macro, as seen on the right image.

Is there already an option in FABulous to distribute the clock to the individual macros in a star shape? Thanks for your help!

mole99, 2024-02-22:
Thanks! I wasn't aware of that, there is still much to learn :D
Excited to see the work of that in nextpnr!

I'm closing this issue as my question has been answered.

PS: Have you thought about setting up a Matrix channel for FABulous? It could be a great place to ask such questions casually. For example [apicula](https://github.com/YosysHQ/apicula) and [coriolis](https://github.com/lip6/coriolis) both have their own Matrix channel.

closed 2024-02-22 by mole99

KelvinChung2000, 2024-02-22:
Unlikely. However,  I do think we need more transparency in the project development and information/knowledge exchange. Probably will need to wait until someone who is going to maintain this full-time.