FPGA-Research/FABulous #201: Switch-Matrix contains commented-out ports

hoyer-ims, 2024-06-27:
Trying to integrate some custom-tiles, I have used the existing tiles and just commented out some ports in the .v of the new BEL with /* */ and //. I have adjusted the switch_matrix.list and .csv for the BEL and removed all of the obsolete ports. However, they still appear in the switch_matrix.csv after FABulous has run, triggering a warning as well: 

[WARNING]-2024-06-27 10:01:12,999 - Input port C15 of switch matrix in Tile MS is not used
[WARNING]-2024-06-27 10:01:12,999 - Input port C16 of switch matrix in Tile MS is not used

I am not sure whether the issue impacts the overall flow of FABulous or whether it is just a thing that can be ignored. It took me a while to figure out that it was the commented-out ports, that caused the issue. Deleting the commented-out part fixed it. 

It can easily be reproduced by adding a commented-out additional port to e. g. the MULADD.v
[image]
[image]

hoyer-ims, 2024-06-27:
Naming ports double in the BEL.v (one time commented-in, one time commented-out) results in no warning, but in an issue when nextpnr tries to route. It gives an assertion failure: 

terminate called after throwing an instance of 'nextpnr_generic::assertion_failure'
  what():  Assertion failure: bi.pins.count(name) == 0 (/home/hoyer/GIT/Fabulous/work/fabulous_efpna/nextpnr_efpna/generic/arch.cc:147)
[ERROR]-2024-06-27 10:53:43,504 - Placement and Routing failed.

KelvinChung2000, 2024-06-27:
Only commenting will keep the port in because the current implementation uses a regular expression to find the port, and commenting will not work. Hopefully, we will use an HDL parser to extract information soon.

Whether a write is double depends on the `.csv` file using the offset attribute. You should be able to name it anything.

closed 2025-01-27 by KelvinChung2000