FPGA-Research/FABulous #204: Issue with io_oeb in User Design

IAmMarcelJung, 2024-07-02:
In the verilog example user design (`sequential_16bit_en.v`), the value assigned to `io_oeb` is `28'b1`.
However, if the resulting bitstream is uploaded to a fabricated eFPGA, this will cause that the output is not working. It will only work if `~(28'b1)` is assigned.

@biswajitks1 and I were investigating why this is the case and we also got help from @everythingelsewasalreadytaken.

In `top_wrapper.v`, `io_oeb`  is passed to the input `T` of  `IO_1_bidirectional_frame_config_pass.v`. In this file `T` is negated and assigned to the output `T_top`. This `T_top` is marked as `EXTERNAL` and is directly connected to the IO tri-state buffer as far as we understood. Although we are not familiar with the PDK, we found four tri-state buffers through the search in the [documentation](https://skywater-pdk.readthedocs.io/en/main/index.html). They all seem to be similar (presumably just different speed grades), so here is one [example](https://skywater-pdk.readthedocs.io/en/main/contents/libraries/sky130_fd_sc_hdll/cells/ebufn/README.html#sky130-fd-sc-hdll-ebufn). There is a 'TE_B' pin, which then would again negate the signal.

In short this means that a '1' is negated two times (once in  `IO_1_bidirectional_frame_config_pass.v`, once in the PDK buffer), so essentially it still has the value '1' when controlling the tri-state inside the PDK buffer. Therefore, unlike the name suggests, 'io_oeb' in the user design is not negating the signal.

The question now is, if this assumption is correct. If the assumption is correct, there should be some changes made either to the signal name and the value or to the negation in  `IO_1_bidirectional_frame_config_pass.v` or maybe there are also more options.

We hope the explanation was somewhat clear. If there is anything unclear, feel free to ask.

KelvinChung2000, 2024-07-02:
This could be due to the IO pin's buffer being inserted manually during the fabrication flow, which is not happening within FABulous and causes the simulation to not match the real chip. I am unsure if this is causing what you see; you will need to ask Dirk about the details. 

I think it is still fine as of now since we have not implemented fabrication integration with FABulous, but later, we will need to clarify it in the documentation.

IAmMarcelJung, 2024-07-02:
Thanks for your quick reply! I will ask @dirk-koch then, also in person and keep this updated with anything that might be discussed then.

KelvinChung2000, 2024-07-02:
We should rename the `io_oeb` to something else, like `out_enable`, which will be more descriptive than `oeb`. Another thing that might be possible is we might be using an uncommon convention of `1` meaning a pin in input as a and `0` means the pin is output since `1nput` and `0utput.`

IAmMarcelJung, 2024-07-02:
> We should rename the `io_oeb` to something else, like `out_enable`, which will be more descriptive than `oeb`.

I personally am also a fan of more descriptive names. They might take up a bit more space, but it is way easier for anyone reading the code to actually know what it means. And `io_oeb` could also be ambigous and mean both `output enable buffer` and `output enable bar` changing it to something like `out_enable` would also fix that.

IAmMarcelJung, 2024-07-05:
So I briefly asked Dirk and he also didn't know an answer right away, but I also didn"t give him the full details. He is quite busy at the moment, so I will ask him again at another time. I will post an update once we had a more in-depth talk.

IAmMarcelJung, 2025-01-06:
I will close this since it is no really a FABulous issue but rather dependent on the technology used for the fabric implementation.

closed 2025-01-06 by IAmMarcelJung