FPGA-Research/FABulous #205: No such file or directory: 'yosys'

CarlosQbit, 2024-07-02:
Hello,
I am facing the following issue, I executed the following lines:

```
docker build -t fabulous .
docker run -it -v $PWD:/workspace fabulous
export FAB_ROOT='workspace'
load_fabric
run_FABulous_fabric
```
and it works fine but when I execute the following line:

`run_FABulous_bitstream npnr ./user_design/sequential_16bit_en.v`

it says:` FileNotFoundError: [Errno 2] No such file or directory: 'yosys'`, the whole error is shown in the attached picture.

[image]

I don't understand this mistake since I am using Docker so everything should be working automatically. I would be grateful if anyone could help me solve this issue. Thanks in advance!

KelvinChung2000, 2024-07-02:
If you look at the dockerfile, you should see yosys and nextpnr is not installed. We have not checked nor updated it for a while. If you need further assistance in installing them please let me know.

CarlosQbit, 2024-07-02:
Thanks for your response. Yes, I would like to receive assistance because I am also not quite familiar with linux and docker.Thanks in advance!

KelvinChung2000, 2024-07-02:
Are you using a Windows device at the moment? I suggest using [wsl](https://canonical-ubuntu-wsl.readthedocs-hosted.com/en/latest/guides/install-ubuntu-wsl2/), which will allow you to have a complete Linux system running under Windows. 

After that, you can follow the installation guide of [yosys](https://yosyshq.readthedocs.io/projects/yosys/en/latest/getting_started/installation.html#build-prerequisites) and [nextpnr-generic](https://github.com/YosysHQ/nextpnr?tab=readme-ov-file#nextpnr-generic) to get rest of the toolchain installed.

CarlosQbit, 2024-07-02:
I am using Linux Ubuntu 20.04 natively, is it the same?

KelvinChung2000, 2024-07-02:
It is the same. You can follow the install guide to install the software.

CarlosQbit, 2024-07-02:
by doing this I will not need docker? or somehow i will install it inside docker?

KelvinChung2000, 2024-07-02:
You will not need to use Docker. You can directly install them on your system. 

Think of Docker as a mini operating system that isolates all your installs from your main system. If you already have Ubuntu running, you will not need to install the software with Docker (sometimes you might still want, but it probably will not apply to you for now).

CarlosQbit, 2024-07-03:
thanks! what should I enter here?
[image]

IAmMarcelJung, 2024-07-03:
In your case, you have to enter `export FAB_ROOT=~/FABulous`, since `FAB_ROOT` has to be set to the root of your FABulous install directory. Note that this will only apply to your current terminal and you have to retype it in every terminal you open. If you want to set it permanently, add `export FAB_ROOT=~/FABulous` to your `~/.bashrc`.

IAmMarcelJung, 2024-07-04:
Since you managed to get into the FABulous shell as mentioned in #207, I suppose the problem described here is solved? If so, please close the issue. If not, let us know how we can further help you!

closed 2024-07-08 by CarlosQbit