FPGA-Research/FABulous #206: LocalWriteStrobe output from eFPGA_Config

RobB720, 2024-07-03:
Hi,

The LocalWriteStrobe output from eFPGA_Config doesn't seem to connect to any other blocks in my generated fabric and in the demo. Is this correct?

Best regards,
Rob.

KelvinChung2000, 2024-07-03:
It is instantiated under the top wrapper. 

https://github.com/FPGA-Research-Manchester/FABulous/blob/eae6d7d6ad58c136857b99702c6275fb7bc6795e/fabric_generator/fabric_gen.py#L1837-L1863

RobB720, 2024-07-03:
Yes it is an output from eFPGA_config instanciated in eFPGA_top, but it doesn't connect to anything, unless I am missing something?

Seems like a redundant signal, but want to make sure!

KelvinChung2000, 2024-07-03:
I misread what you said and commented, which I found out later, so I deleted the comment. 

My answer is I am not sure and will need to ask @dirk-koch. But since we did not use it and could simulate without it, it is probably unnecessary?