FPGA-Research/FABulous #207: ERROR: No such command: synth_fabulous (type 'help' for a command overview)

CarlosQbit, 2024-07-03:
Hi, I am runnning the example from the documentation like this:

```
carlos@carlos-quantum-pc:~$ cd FABulous
carlos@carlos-quantum-pc:~/FABulous$ export FAB_ROOT=~/FABulous
carlos@carlos-quantum-pc:~/FABulous$ python3.9 FABulous.py -c example_native3
carlos@carlos-quantum-pc:~/FABulous$ python3.9 FABulous.py example_native3
load_fabric

```
and it works fine, but when I run the following line:

`run_FABulous_bitstream npnr ./user_design/sequential_16bit_en.v
`
it drops the next error:

[image]

Someone already had this problem and they told him to reinstall yosys. I have already done that but the issue persists. Thanks in advance!

IAmMarcelJung, 2024-07-04:
I previously also had the same problem. You probably have the wrong Yosys verson. Can you enter `yosys --version` and show the output? How did you install it? IIRC, I got the issue when I installed it using `apt`. In the installation manual they mention that Yosys is included in the OSS CAD suite, which also comes with many more useful tools, so I just installed that and then it worked. You could also build Yosys from source but just installing the OSS CAD suite is probably easier, since you alredy mentioned that you not yet used to Linux.

Let us know if that helped or if you need further assistance :)

IAmMarcelJung, 2024-07-04:
I just checked the [apt versions](https://packages.ubuntu.com/search?keywords=yosys) and if you installed it on Ubuntu 20.04, then you will have version 0.9-1build2, which is an older version than the minimum required one as mentioned in the [documentation](https://fabulous.readthedocs.io/en/latest/Usage.html)  (0.26+0). The next LTS of Ubuntu, 24.04, will have a way more recent version which would then already include the `synth_fabulous` command.

CarlosQbit, 2024-07-05:
I will update Ubuntu and then tell you what happened

CarlosQbit, 2024-07-06:
Now when running the following line:
`run_FABulous_bitstream npnr ./user_design/sequential_16bit_en.v `
it does generate the bitstream but there appears an error that I do not know if it affects the result. Also, I tried uninstalling and installing fasm as recommended there, as well as these dependencies `pip3 install -r requirements.txt`
but I also don't know if they're the dependencies the terminal is talking about. 
[image]

Regardless of that, I run the simulation with
`./build_test_design.sh`
and then
`./run_simulation.sh`
and get this, which I am not sure if it means it run correctly. Thanks in advance

IAmMarcelJung, 2024-07-06:
Congrats! Everything works as expected! :) The import error was fixed in #110 IIRC, but this is not yet merged into master. The error does not affect the generated FASM and Bitstream file. 

It should be noted that `build_test_design.sh` does not actually test the user design, but a separate test design. If you want to run the simulation for the user design, you can use the `run_simulation` command, which still has to be mentioned in the documentation. We should also think about printing some success message if the simulation ran sucessfully.

The simulation checks the fabric IO signals (`I_top`, `T_Top`) against the output of the user design (`gold`). If they are equal (as in your case), the simulation succeeded! If they differ, the simulation throws an error. 

Just out of curiosity, which method did end up using for the Yosys installation?

CarlosQbit, 2024-07-08:
Thanks for your support!
I installed using:
sudo apt-get update
sudo apt-get -y install yosys
Thanks for all your help, I will keep on making experiments!

CarlosQbit, 2024-07-08:
Well now I do have a quesion. Is there a way to see the waveform?

KelvinChung2000, 2024-07-08:
You can use [GTKWave](https://zoomadmin.com/HowToInstall/UbuntuPackage/gtkwave). I personally use [Surfer](https://gitlab.com/surfer-project/surfer), which can be a standalone program or can be used within VSCode as an [extension](https://marketplace.visualstudio.com/items?itemName=surfer-project.surfer)

IAmMarcelJung, 2024-07-10:
> Thanks for your support!

You're welcome! :)

> I installed using:
> sudo apt-get update
> sudo apt-get -y install yosys

So you upgraded to Ubuntu 24.04? Since all other Ubuntu versions use too old versions of Yosys if I got that correctly.

> Thanks for all your help, I will keep on making experiments!

Have fun with your experiments, and don't hesitate to ask more questions! :)

CarlosQbit, 2024-07-19:
Yes I installed Ubuntu24.04 :)

Thanks to both of you!

closed 2024-07-19 by CarlosQbit