FPGA-Research/FABulous #213: Waveform

CarlosQbit, 2024-07-29:
Hello, I hope you are having a great day. I am trying to compare the waveforms of the test circuit (I found out it was a counter) with the waveform of the ouput of the eFPGA after simulation, after following all the Quick Start Documentation and Simulation Setup Documentation:

```
python3 FABulous.py -c <name_of_project>
python3 FABulous.py <name_of_project>
# inside the FABulous shell
load_fabric
run_FABulous_fabric
run_FABulous_bitstream npnr user_design/sequential_16bit_en.v

cd demo/Test
./build_test_design.sh
./run_simulation.sh
```

 I did what is on the picture to generate the waveform because otherwise I could not find how to. iI would be glad if someone could explain if there is a better way to generate the waveform and compare it. Also if there is some documentation or an explanation on how to try a different circuit besides the implemented counter, for examle an AND gate or a multiplier.  Also if there is a way to see the waveform of the bitstream. Thanks in advance!
[image]

KelvinChung2000, 2024-07-29:
Since we are using Icarus, this is the only way to generate a waveform. If you are trying to compare different waveforms, the only thing I am aware of is a relatively recent paper on waveform comparison language tools, which I am not familiar with. The main problem is that two waveforms can be different but perform the same thing. Furthermore, the waveform generated by different simulators will be different as well. What might be more practical is to have the simulator print value out and do the comparison that way. 

The simplest thing you can do is keep the counter's interface and replace the internal logic with what you want to experiment with. The current waveform you are getting also contains the bitstream loading action. 

Based on the type of question you are asking, you have no previous experience in RTL design. If you are planning to use FABulous for your research, email me at king.chung@manchester.ac.uk, and we can see how to better assist you.

IAmMarcelJung, 2024-07-30:
Hi, I don't know how much you already solved in personal communication, but I'll just leave my two cents here:

> I am trying to compare the waveforms of the test circuit [...] with the waveform of the ouput of the eFPGA after simulation [...].

You can do so directly in the waveform generated by the fabric testbench `fabulous_tb.v`. Since the whole fabric is simulated including the user design, you can view both the fabric output and the user design output in the same waveform. The simulation console output also compares the output of the counter to the fabric ouput, as can be seen when looking for the source of the `top` and `top_gold` signals in the module instantiations (thanks to @Biswajitks1 for pointing this out recently):

```verilog
    // Instantiate both the fabric and the reference DUT
    eFPGA_top top_i (
        .I_top(I_top),
        .T_top(T_top),
        .O_top(O_top),
        .A_config_C(A_cfg), .B_config_C(B_cfg),
        .CLK(CLK), .resetn(resetn),
        .SelfWriteStrobe(SelfWriteStrobe), .SelfWriteData(SelfWriteData),
        .Rx(Rx),
        .ComActive(ComActive),
        .ReceiveLED(ReceiveLED),
        .s_clk(s_clk),
        .s_data(s_data)
    );

    wire [27:0] I_top_gold, oeb_gold, T_top_gold;
    top dut_i (
        .clk(CLK),
        .io_out(I_top_gold),
        .io_oeb(oeb_gold),
        .io_in(O_top)
    );

```
Remember that the design used in the script is in  `$(project_name)/Test/test_design/counter.v` and the module is named `top`.
Here you can see the waveform comparison:

[image]

> I did what is on the picture to generate the waveform because otherwise I could not find how to.

The part you added was this, right?:

```verilog
        $dumpfile("waveform.vcd");
        $dumpvars();
```
I think the `.fst` generation is currently broken and some commands need more flags, so I will look into this (Currently I think that just `vvp` needs the `-fst` flag). That's why you would currently need the `.vcd` option you added.

After running the simulation, `waveform.vcd` should be generated. You can open it with a waveform viewer of your choice that supports the `.vcd` format as mentioned by @KelvinChung2000  in #207.

>Also if there is some documentation or an explanation on how to try a different circuit besides the implemented counter, for examle an AND gate or a multiplier

There is also the `run_simulation` command in the FABulous shell, which directly simulates the user design. This still needs to be documented (reminder to myself...). But you can use `help` to get at least some information about it for now. You could of course also copy your design into the `test_design` and adjust the tes setup to it or place the module as the top module in `counter.v`. But consider that not the design itself is tested, but rather the fabric itself (which does not mean that it can still be useful).

 >Also if there is a way to see the waveform of the bitstream

In the simulation, the bitstream is loaded into the configuration storage, so the waveform you see already reflects the bistream. Just be sure that you are comparing the correct bitstream to the correct user design.

I hope I could clarify some things and did not tell you too much you already knew. As always, feel free to ask further questions!

closed 2025-01-27 by KelvinChung2000