FPGA-Research/FABulous #228: Using python3.10 feature but python3.9 is specified in README.md (FABulous2.0-development branch)

IAmMarcelJung, 2024-08-20:
When I wanted to test the changes in #227, I saw that the python version to use is specified as `python3.9` in the `FABulous2.0-development`  branch. Therefore I set my venv to `python3.9` and set it up as usual. When I wanted to create the project, using `FABulous -c demo` I got the following error (shortened):

```console
File "FABulous/fabric_definition/Fabric.py", line 249, in Fabric
    def getTileByName(self, name: str) -> Tile | None:
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
```
I then found out that the `union` type was introduced in [PEP 604](https://peps.python.org/pep-0604/) and is only supported in `python3.10` and higher. Therefore I suggest we either remove the union type or change the suggested python version in `README.md` to at least `python3.10` in the `FABulous2.0-development` branch (which should be preferred IMHO).

If there is an error on my side please let me know, but I think the PEP is quite clear.

KelvinChung2000, 2024-08-20:
We might as well move to python3.12 since our CI is currently running at python3.12.

IAmMarcelJung, 2024-08-20:
Yeah, then we should specify `python3.12` in `README.md` and also in `pyproject.toml` (thanks to @EverythingElseWasAlreadyTaken for the hint). I will create a PR for this.

IAmMarcelJung, 2024-08-20:
> We might as well move to python3.12 since our CI is currently running at python3.12.

For me it looks like we still use [3.10 in the CI](
https://github.com/FPGA-Research-Manchester/FABulous/blob/FABulous2.0-development/.github/workflows/fabric_gen.yml#L13-L16). Or am I looking at the wrong file? Just want to make sure before I change the docs to 3.12 :)

KelvinChung2000, 2024-08-20:
I thought it is on 3.12🤔. Anyway, I think we should still move to 3.12.

For me it looks like we still use 3.10 in the CI [github.com]<https://urldefense.com/v3/__https://github.com/FPGA-Research-Manchester/FABulous/blob/FABulous2.0-development/.github/workflows/fabric_gen.yml*L13-L16__;Iw!!PDiH4ENfjr2_Jw!HQUaTyWnYqkdHIEAXZpSJObahQ4rPmFAGmyfPb_I_Gqwwkvrkw25MkGGcugHlpSU4TdMlI4p2NE_BJjJoRzcrnZKrLbWE1lI$>. Or am I looking at the wrong file? Just want to make sure before I change the docs to 3.12 :)

IAmMarcelJung, 2024-08-20:
No problem! So should I change everything to 3.12? Also for the docs and the CI?

KelvinChung2000, 2024-08-20:
Yes, please.

PR #229 (2024-08-22): Update python to 3.12 and update documentation
This fixes #228 by changing the required python version to 3.12. Since `Usage.rst` had to be updated, small improvements were also made to that file which are listed below:

- The note about the fasm warning was hidden and is now visible.
- The subsection headings for the docker section were wrapped into two lines and are now changed to just bold text. They could have also be changed to an actual heading, but the next heading level was too big compared to the previous solution using fields and the level seem to be managed automatically in restruct [... cut]