FPGA-Research/FABulous #230: Persistent command line history not working on FABulous2.0 branch

IAmMarcelJung, 2024-08-28:
When testing the edits made in the recent pull requests, I noticed that the persistent CLI history (being able to recall commands from previous FABulous sessions) introduced in #132 did not work anymore in the FABulous2.0 branch. It still does work for me in the master branch. Can anyone confirm or deny this behaviour? If it can be confirmed, was it removed on purpose or by accident?

Thanks in advance for any help!

KelvinChung2000, 2024-08-28:
I have experienced the same, and it is likely by accident.

KelvinChung2000, 2024-08-28:
When I am working on the release, I will spend some time using LLM to generate tests for the software.

IAmMarcelJung, 2025-02-07:
This is now fixed, probably by #273.

closed 2025-02-07 by IAmMarcelJung