FPGA-Research/FABulous #236: Dependency Revaluation

mole99, 2024-10-02:
Hello! I would like to package FABulous with Nix, but there are two dependencies that complicate things. They are listed in `pyproject.toml` as `fasm` and `docker` (I am looking at `FABulous2.0-development`).

As far as I can see, the [`docker`](https://pypi.org/project/docker/) Python package is never used in FABulous. Maybe the Python library was confused with the Docker engine, which is indeed necessary to build and run the FABulous docker container? If so, could this dependency be removed?

With `fasm` it's a bit more complicated as it is used to generate the bitstream. From what I see `fasm` is imported in two files: `fabric_gen.py` and `bit_gen.py`. There is a note at each import: `# Remove this line if you do not have the fasm library installed and will not be generating a bitstream`. Would it be possible to import `fasm` with a `try / except`? This would make it possible to run FABulous without `fasm` installed and at least generate a fabric.

Another question, does `fabric_gen` actually make use of fasm? I don't see any functions being used. It would be helpful if instead of `from fasm import *` only selected functions could be imported `from fasm import ..., ...` this makes it easier to see which functions are actually used.

On a side note, I also don't see `numpy` being used in FABulous. Since it is another huge dependency, maybe it could be removed too? Lastly, I think `python-dotenv` and `loguru` should be added to `pyproject.toml`.

I would be happy to help with any changes!

KelvinChung2000, 2024-10-02:
The `docker` dependency is related to something I worked on but did not release, so it is safe to remove. 

I know that `fabric_gen.py` should not be using `fasm` so this should be safe to remove. I think `fasm` is not being used by `bit_gen.py` as well and have consider removing it but I cannot confirm it is not being used. The `from fasm import *` is also from pre-refactor. The `bit_gen.py` file should be removed and changed to fit the new software architecture...   

The `numpy` package is something from pre-refactor, which I don't think is currently being used.  

`python-dotenv` and `loguru` should be added to the `pyproject.toml`.

I have this discussion of migrating towards Nix for better Openlane2 integration; I don't know if this is being worked on at the moment. Ultimately, we aim to support driving the Openlane2 flow from FABulous.

EverythingElseWasAlreadyTaken, 2024-10-02:
It would also be nice to add required package versions in the `requirements.txt` and `docs/requirements.txt`.

mole99, 2024-10-02:
@KelvinChung2000 Well, I'm working on the opposite right now :) Driving FABulous from OpenLane 2. 
For that I created an OL2 plugin with a flow called `FABulousTile` to generate the layout for the tiles and another flow called `FABulousFabric` to stitch the final fabric. Thus, a simple OL2 config file is enough to start the flow.

@EverythingElseWasAlreadyTaken Yes that is a good idea, but please don't pin exact versions or packaging will be difficult.

KelvinChung2000, 2024-10-02:
I know you are working on it, and I have read the paper. I would like to understand why you would like to drive FABulous by Openlane. Since FABulous is a design tool, it makes more sense to me that a design drives the implementation than an implementation drives the design.

mole99, 2024-10-09:
I have now created a PR with the changes in the dependencies: #239 

Unfortunately, `fasm` could not be completely removed as it is used for bitstream generation. If I remove the import in `bit_gen.py`, I get the following error:

```
NameError: name 'parse_fasm_filename' is not defined
```

If it would be possible to remove fasm in the future, that would definitely help to create a fully featured Nix package. For now, I just leave `fasm` out of the dependencies as I don't use the package to create a bitstream, only to generate the fabric.

IAmMarcelJung, 2025-02-07:
Most of this is fixed in #239, do we want to keep this open since `fasm` is still in there or should we close this?

EverythingElseWasAlreadyTaken, 2025-05-28:
I think this issue is mostly resolved, and we currently can't get rid of fasm. 
But this could change if we do #330.

I'll close this issue, please reopen if I missed something here.