FPGA-Research/FABulous #245: Difference in bitstream generation between master and dev branch

IAmMarcelJung, 2024-10-29:
Hi everyone,

when playing around with the MPW-5 FABulous chips, I noticed that the bitstream generated by the current version of the `dev` branch differs from the one created by the `master` branch. Also, the bitstream created by the `dev` branch version does not work as expected. I cloned the repo to a new location to make sure I have the most recent version of the branches and no old leftovers or anything. I also used the default user design to make sure I didn't make any changes. Still there could of course be some other configuration issue on my end.

Therefore I would like to know if anyone can either deny or confirm behaviour I'm experiencing.

Thanks in advance!

KelvinChung2000, 2024-10-29:
I suggest checking the generated bitstream spec file. The master and dev should match. The `bitStreamSpec.bin` is a pickle, so you can load that into Python to compare. 

But this is relatively strange since the simulation is passing.

IAmMarcelJung, 2024-10-29:
Both `bitStreamSpec.bin` and `bitStreamSpec.csv` do not match on the two branches (which makes sense since they are created from the same data). I looked at the differences in `bitStreamSpec.csv`, but there were a lot of them. The simulation is also executed in the CI pipeline, right?

KelvinChung2000, 2024-10-29:
The simulation is executed in the CI pipeline. 

If you create two of the same fabrics, one with the master and one with the dev, the key-value pair of the pickled dict should match. The CSV is less important since it is for human reading. The `bitStreamSpec.bin` is used in the simulation to generate the bitstream.

IAmMarcelJung, 2024-10-29:
>  the key-value pair of the pickled dict should match
So it is okay that the files are not exactly the same? Only the pairs have to match?

KelvinChung2000, 2024-10-29:
> > the key-value pair of the pickled dict should match
> 
> So it is okay that the files are not exactly the same? Only the pairs have to match? Which pairs do you mean exactly?

Yes, the order of the entry might change, but the key-value pair should match if I remember correctly. 

> I executed the simulation for both cases and also found out that both are working, although the `bitstream.hex` file and therefore the `SelfWriteData` also differs. Maybe the design is simple enough so there is no effect from the difference?

I assume they should have the same configuration logic. Is the order of bitstream different? like little endian and big endian? I am not sure about the details of the bitstream and configuration.

IAmMarcelJung, 2024-10-29:
>  Is the order of bitstream different?

No the order is the same for the simulation counter, only some bytes differ (20-30). 

Here are the `.hex` files used in the simulation script if you want to look for yourself (renamed to `.txt` so GitHub accepts the files):
[bitstream_master.txt](https://github.com/user-attachments/files/17559328/bitstream_master.txt)
[bitstream_dev.txt](https://github.com/user-attachments/files/17559347/bitstream_dev.txt)

I'm currently fine with using the `master` branch, but I think before we are merging the `dev` branch we should be sure that there is acutally no difference in the bitstream generation (so if anyone can just deny my findings that would already be fine). I also looked in `FABulous.py` and `bit_gen.py` but couldn't see any changes that might affect the bitstream generation so this is still quite strange.

IAmMarcelJung, 2025-04-15:
Just checked in hardware that #352 fixes this. Thanks @KelvinChung2000!

As a sidenote, the bitstreams still differ, but this is just due to some reordering which I checked in a diff.

closed 2025-04-15 by IAmMarcelJung

KelvinChung2000, 2025-04-15:
Is the bistream spec different, or is the bistream different?

IAmMarcelJung, 2025-04-16:
If you create a project with the version of the `master` branch and let the flow run through both the `master` and the `dev` branch version it's the same. Only if you compare a project created and run on the `master` branch  version with a project created and run on the `dev` branch, both the spec and the bitstream itself differ.

So depending on how you see it, there either still is a difference or not...

KelvinChung2000, 2025-04-16:
This is fine then. This is "forward incompatible".