FPGA-Research/FABulous #249: Buffer in tiles

Sacoder0000, 2024-11-13:
I noticed a lot buffers are used for driving FrameData and FrameStrobe. I am wondering that the synthesis tools will automatically add buffers, why do we add them in .v file?

KelvinChung2000, 2024-11-13:
I assume you are talking about the eFPGA ASIC implementation. Those buffers must be added manually to ensure signal integrity.

Sacoder0000, 2024-11-13:
Thank you for replying. Yes, I am talking about eFPGA ASIC synthesis and implementation. I am now using Design Compiler and IC Compiler to implement this projects "https://github.com/FPGA-Research-Manchester/eFPGA---RTL-to-GDS-with-SKY130", and I think signal integrity will be automatic fixed by implementation tool. 
[image]
[image]

Sacoder0000, 2024-11-13:
> I assume you are talking about the eFPGA ASIC implementation. Those buffers must be added manually to ensure signal integrity.

I think buffers will be added during synthesis optimazition and implementation. I am still new to ASIC physics implementation,  so please don't mind if I say anything wrong.

KelvinChung2000, 2024-11-13:
I am not an expert in ASIC implementation, and I am not the main person working on the ASIC. I know tools can do automatic buffer insertion. It could be a difference in the tools we use. We mainly use the Openlane flow, which is relatively linear; I just checked, and they have just added automatic buffer insertion (alpha stage).

Sacoder0000, 2024-11-13:
Thank you for your reply!
So, I will try to remove buffers in FABulous output .v files and see what happen.

KelvinChung2000, 2024-11-14:
After talking to someone more knowledgeable, I learned that the buffer is mainly for cases where you harden each tile and stitch them together. When stitching the wire, the wire becomes much longer, which requires a buffer. 

If you are trying to compile the design flat, you should be safe to remove the buffer; otherwise, you will want to keep it.

closed 2024-12-07 by Sacoder0000