FPGA-Research/FABulous #251: Supertiles and normal tiles

Sacoder0000, 2024-11-16:
Hello! I am about to design a personal fabric.

What is the fundamental difference between supertile and tile? After reading the FABulous Document, it seems to me that there is only a difference in area? Is it necessary to make a supertile as long as the area is twice or more than the usual tile area? But this raises another problem:

As shown in the fabric.csv layout, it seems that every tile's area and placement is assumed, like the LUT column and Regfile column, which are one-to-one. However, after synthesis, the area of the LUT is much larger than the Regfile(eg:6000-4000umxum), which I think is likely to cause a lot of waste of area.
[image]

KelvinChung2000, 2024-11-16:
Supertiles are for Bels that needs more wire than what one switch matrix can provide. 

Yes, LUT tile will be much larger in size. So what you want to do is first synthesis the LUT tile to as small as possible. Then you synthesis the other tile with a fixed height that is same as the LUT tile, and try to minimise the width of the tile.

Sacoder0000, 2024-11-16:
Couldn’t I synthesis them together? with restrict SDC constraints?

Sacoder0000, 2024-11-16:
I will try comparing the strategies of synthesizing one by one and synthesizing all together to see which one works better.

KelvinChung2000, 2024-11-16:
I am not sure what SDC can do. But by ensuring the LUT is smallest gives you better overall density.

Sacoder0000, 2024-11-16:
Thanks for your advice, SDC is a timing constraints file, which i used to break timing loops in FPGA and define the frequency of CLK used in FPGA during synthesis.

KelvinChung2000, 2024-11-16:
You can try give this paper a read https://dl.acm.org/doi/10.1145/3637543.3652874 which compares synthesis the design flat vs synthesis the design with hierarchy.

Sacoder0000, 2024-11-16:
Wow, that's really a big help!! I needed this paper so much!!!!

closed 2024-12-07 by Sacoder0000