FPGA-Research/FABulous #283: Full support for bit vectors in User BELs

mole99, 2025-01-23:
Thanks to #222, the User BEL modules have support for bit vectors. Unfortunately, this is not yet supported in the complete flow.

Once a User BEL is instantiated inside a tile, the bit vectors are split up into individual bits:

<details>
  <summary>BEL Instantiation</summary>

  ```Verilog
   //BEL component instantiations
EF_SRAM_1024x32 Inst_EF_SRAM_1024x32 (
    .AD({AD9, AD8, AD7, AD6, AD5, AD4, AD3, AD2, AD1, AD0}),
    .BEN({BEN31, BEN30, BEN29, BEN28, BEN27, BEN26, BEN25, BEN24, BEN23, BEN22, BEN21, BEN20, BEN19, BEN18, BEN17, BEN16, BEN15, BEN14, BEN13, BEN12, BEN11, BEN10, BEN9, BEN8, BEN7, BEN6, BEN5, BEN4, BEN3, BEN2, BEN1, BEN0}),
    .DI({DI31, DI30, DI29, DI28, DI27, DI26, DI25, DI24, DI23, DI22, DI21, DI20, DI19, DI18, DI17, DI16, DI15, DI14, DI13, DI12, DI11, DI10, DI9, DI8, DI7, DI6, DI5, DI4, DI3, DI2, DI1, DI0}),
    .EN(EN),
    .R_WB(R_WB),
    .DO({DO31, DO30, DO29, DO28, DO27, DO26, DO25, DO24, DO23, DO22, DO21, DO20, DO19, DO18, DO17, DO16, DO15, DO14, DO13, DO12, DO11, DO10, DO9, DO8, DO7, DO6, DO5, DO4, DO3, DO2, DO1, DO0}),
    .DO_SRAM({DO_SRAM31, DO_SRAM30, DO_SRAM29, DO_SRAM28, DO_SRAM27, DO_SRAM26, DO_SRAM25, DO_SRAM24, DO_SRAM23, DO_SRAM22, DO_SRAM21, DO_SRAM20, DO_SRAM19, DO_SRAM18, DO_SRAM17, DO_SRAM16, DO_SRAM15, DO_SRAM14, DO_SRAM13, DO_SRAM12, DO_SRAM11, DO_SRAM10, DO_SRAM9, DO_SRAM8, DO_SRAM7, DO_SRAM6, DO_SRAM5, DO_SRAM4, DO_SRAM3, DO_SRAM2, DO_SRAM1, DO_SRAM0}),
    .AD_SRAM({AD_SRAM9, AD_SRAM8, AD_SRAM7, AD_SRAM6, AD_SRAM5, AD_SRAM4, AD_SRAM3, AD_SRAM2, AD_SRAM1, AD_SRAM0}),
    .BEN_SRAM({BEN_SRAM31, BEN_SRAM30, BEN_SRAM29, BEN_SRAM28, BEN_SRAM27, BEN_SRAM26, BEN_SRAM25, BEN_SRAM24, BEN_SRAM23, BEN_SRAM22, BEN_SRAM21, BEN_SRAM20, BEN_SRAM19, BEN_SRAM18, BEN_SRAM17, BEN_SRAM16, BEN_SRAM15, BEN_SRAM14, BEN_SRAM13, BEN_SRAM12, BEN_SRAM11, BEN_SRAM10, BEN_SRAM9, BEN_SRAM8, BEN_SRAM7, BEN_SRAM6, BEN_SRAM5, BEN_SRAM4, BEN_SRAM3, BEN_SRAM2, BEN_SRAM1, BEN_SRAM0}),
    .DI_SRAM({DI_SRAM31, DI_SRAM30, DI_SRAM29, DI_SRAM28, DI_SRAM27, DI_SRAM26, DI_SRAM25, DI_SRAM24, DI_SRAM23, DI_SRAM22, DI_SRAM21, DI_SRAM20, DI_SRAM19, DI_SRAM18, DI_SRAM17, DI_SRAM16, DI_SRAM15, DI_SRAM14, DI_SRAM13, DI_SRAM12, DI_SRAM11, DI_SRAM10, DI_SRAM9, DI_SRAM8, DI_SRAM7, DI_SRAM6, DI_SRAM5, DI_SRAM4, DI_SRAM3, DI_SRAM2, DI_SRAM1, DI_SRAM0}),
    .EN_SRAM(EN_SRAM),
    .R_WB_SRAM(R_WB_SRAM)
);
  ```
  
</details>

There are two issues with that:

1. If the BEL contains external I/Os, integrating the fabric into a larger system is very cumbersome.
2. The primitive exported to nextpnr also uses individual bits.

Instead of simply instantiating the primitive inside the user design as:

```
EF_SRAM_1024x32_wrapper i_EF_SRAM_1024x32 (
        .AD     (...),
        .BEN    (...),
        .DI     (...),
        .DO     (...),
        .EN     (...),
        .R_WB   (...)
    );
```

I first need to create a wrapper to be able to handle all of the bits more easily:

<details>
  <summary>Primitive Instantiation</summary>

  ```Verilog
module EF_SRAM_1024x32_wrapper (
    input  [ 9:0] AD,
    input  [31:0] BEN,
    input  [31:0] DI,
    input         EN,
    input         R_WB,
    output [31:0] DO
);
    (* keep *) EF_SRAM_1024x32 i_EF_SRAM_1024x32 (
        .AD0    (AD[0]),
        .AD1    (AD[1]),
        .AD2    (AD[2]),
        .AD3    (AD[3]),
        .AD4    (AD[4]),
        .AD5    (AD[5]),
        .AD6    (AD[6]),
        .AD7    (AD[7]),
        .AD8    (AD[8]),
        .AD9    (AD[9]),
        
        .BEN0   (BEN[0]),
        .BEN1   (BEN[1]),
        .BEN2   (BEN[2]),
        .BEN3   (BEN[3]),
        .BEN4   (BEN[4]),
        .BEN5   (BEN[5]),
        .BEN6   (BEN[6]),
        .BEN7   (BEN[7]),
        .BEN8   (BEN[8]),
        .BEN9   (BEN[9]),
        .BEN10   (BEN[10]),
        .BEN11   (BEN[11]),
        .BEN12   (BEN[12]),
        .BEN13   (BEN[13]),
        .BEN14   (BEN[14]),
        .BEN15   (BEN[15]),
        .BEN16   (BEN[16]),
        .BEN17   (BEN[17]),
        .BEN18   (BEN[18]),
        .BEN19   (BEN[19]),
        .BEN20   (BEN[20]),
        .BEN21   (BEN[21]),
        .BEN22   (BEN[22]),
        .BEN23   (BEN[23]),
        .BEN24   (BEN[24]),
        .BEN25   (BEN[25]),
        .BEN26   (BEN[26]),
        .BEN27   (BEN[27]),
        .BEN28   (BEN[28]),
        .BEN29   (BEN[29]),
        .BEN30   (BEN[30]),
        .BEN31   (BEN[31]),
        
        .DI0    (DI[0]),
        .DI1    (DI[1]),
        .DI2    (DI[2]),
        .DI3    (DI[3]),
        .DI4    (DI[4]),
        .DI5    (DI[5]),
        .DI6    (DI[6]),
        .DI7    (DI[7]),
        .DI8    (DI[8]),
        .DI9    (DI[9]),
        .DI10    (DI[10]),
        .DI11    (DI[11]),
        .DI12    (DI[12]),
        .DI13    (DI[13]),
        .DI14    (DI[14]),
        .DI15    (DI[15]),
        .DI16    (DI[16]),
        .DI17    (DI[17]),
        .DI18    (DI[18]),
        .DI19    (DI[19]),
        .DI20    (DI[20]),
        .DI21    (DI[21]),
        .DI22    (DI[22]),
        .DI23    (DI[23]),
        .DI24    (DI[24]),
        .DI25    (DI[25]),
        .DI26    (DI[26]),
        .DI27    (DI[27]),
        .DI28    (DI[28]),
        .DI29    (DI[29]),
        .DI30    (DI[30]),
        .DI31    (DI[31]),
        
        .EN     (EN),

        .R_WB   (R_WB),
        
        .DO0    (DO[0]),
        .DO1    (DO[1]),
        .DO2    (DO[2]),
        .DO3    (DO[3]),
        .DO4    (DO[4]),
        .DO5    (DO[5]),
        .DO6    (DO[6]),
        .DO7    (DO[7]),
        .DO8    (DO[8]),
        .DO9    (DO[9]),
        .DO10    (DO[10]),
        .DO11    (DO[11]),
        .DO12    (DO[12]),
        .DO13    (DO[13]),
        .DO14    (DO[14]),
        .DO15    (DO[15]),
        .DO16    (DO[16]),
        .DO17    (DO[17]),
        .DO18    (DO[18]),
        .DO19    (DO[19]),
        .DO20    (DO[20]),
        .DO21    (DO[21]),
        .DO22    (DO[22]),
        .DO23    (DO[23]),
        .DO24    (DO[24]),
        .DO25    (DO[25]),
        .DO26    (DO[26]),
        .DO27    (DO[27]),
        .DO28    (DO[28]),
        .DO29    (DO[29]),
        .DO30    (DO[30]),
        .DO31    (DO[31])
    );

endmodule
  ```
  
</details>

---

It would be great if FABulous had full support for bit vectors in User BELs. Thanks!

KelvinChung2000, 2025-01-23:
This will require substantial changes in how to generate the fabric. I want to make this possible, which is also part of the original intention, but it will unlikely happen within the next 2-3 months.

mole99, 2025-01-23:
That's great to hear! Yes, I was fully expecting that this might take a bit more time :)