FPGA-Research/FABulous #284: CLI: allow for extra plib and map files

mole99, 2025-01-23:
[synth_fabulous](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/synth_fabulous.html#cmd-synth_fabulous) has support for `-extra-plib` and `-extra-map` to be used for User defined BELs.

The CLI currently has no option to supply those arguments. I marked this as a bug as I think this is an unintentional limitation of the CLI.

---

Side remark: As a user I am not too happy about "FABulous doing everything for me". Synthesizing the user bitstream and running PnR works well enough in an extra Makefile or shell script. Once the FPGA has been designed and taped-out, I would rather not use FABulous just to create a bitstream for a user design.

IAmMarcelJung, 2025-01-23:
> [synth_fabulous](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/synth_fabulous.html#cmd-synth_fabulous) has support for -extra-plib and -extra-map to be used for User defined BELs.

The CLI currently has no option to supply those arguments. I marked this as a bug as I think this is an unintentional limitation of the CLI.

We should probably make all the the options of `synth_fabulous` accessible in the CLI in the future. @KelvinChung2000 once mentioned that it probably easy due to the migration to cmd2 in #273 IIRC.

> Side remark: As a user I am not too happy about "FABulous doing everything for me". Synthesizing the user bitstream and running PnR works well enough in an extra Makefile or shell script. Once the [...]

I think this depends on what the philosophy behind FABulous should be. Do we want it to be just a Fabric Generator, or more a general CAD toolchain with more features and external programs already included?
 IMHO there is a charm in having the whole flow (or rather both flows) in one tool, so that you can both create a fabric and later also generate a bitstream for it. Especially for users that are not too familiar with all the CAD tools. And even if all of this is supported, its also perfectly fine to do it another way, if that fits your need better :) 

If you want to discuss more about this we should probably move that part of the converstation to a discussion (sorry I'm a bit of a organisation maniac...).

KelvinChung2000, 2025-01-23:
We have the CLI first, then have the synth command. So, the synth command does not support those options.

mole99, 2025-01-23:
@IAmMarcelJung

As I was discussing with @EverythingElseWasAlreadyTaken on Tuesday, I'd really prefer if FABulous could be used as a tool instead of driving the flow. When designing an ASIC, generating the fabric and therefore FABulous is just one part of it. And I feel that trying to automate too much will lead to restrictions/limitations in other areas. I already feel this a little when using FABulous for my master's thesis.

Of course being able to use both options is fine if one does not restrict the other too much.

Taking a deeper look into OpenLane 2 and it's configurability shows that it takes a lot to make a flow both easy to use but configurable enough that it does not impose any restrictions on the user.

But like you said, that should probably be its own discussion by itself ;)

EverythingElseWasAlreadyTaken, 2025-01-23:
> [synth_fabulous](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/synth_fabulous.html#cmd-synth_fabulous) has support for `-extra-plib` and `-extra-map` to be used for User defined BELs. [...]

I'm currently working on supporting custom primitives and generating custom tiles out of the box. This includes also the integration in the synth_fabulous flow. This could partly solve your issue, but it will be mostly suitable for simpler designs.

> Side remark: As a user I am not too happy about "FABulous doing everything for me". Synthesizing the user bitstream and running PnR works well enough in an extra Makefile or shell script. Once the [...]

The end-to-end flow, including synth_fabulous, is mostly designed for the general, but simple use case. For example, if you want to design a simple fabric, with only default tiles, you can do it in no time, without even thinking about how to do all the stuff. 

But you don't have to use it that way. If you are doing more sophisticated design, then of course you should write your own flow for your user design. 

We should maybe add some more documentation for the more sophisticated approach.

closed 2025-01-30 by KelvinChung2000