FPGA-Research/FABulous #286: Cleanup: Remove Fabric CAD

mole99, 2025-01-23:
Remove the Fabric CAD directory (`FABulous/FABulous/fabric_cad/`) from this repository.

I believe it is not up to date anymore and obsolete since the files are hosted on yosys side now: https://github.com/YosysHQ/yosys/tree/main/techlibs/fabulous

This is already reflected in the documentation: https://fabulous.readthedocs.io/en/fabulous2.0-development/FPGA_CAD-tools/yosys.html

IAmMarcelJung, 2025-01-23:
Thanks for pointing this out (and all the other stuff, you just did/are currently doing)!  :) The repo needs indeed some cleanup also in some other places.

EverythingElseWasAlreadyTaken, 2025-01-23:
Yes, you are both are right, this is some legacy stuff we don't use anymore.

I'll open a draft PR, for a FABulous cleanup, where we can collect all the stuff.

IAmMarcelJung, 2025-02-07:
Fixed by #290.

closed 2025-02-07 by IAmMarcelJung