FPGA-Research/FABulous #287: Support PCF for I/Os

mole99, 2025-01-23:
As far as I can see, the current user design uses a [wrapper](https://github.com/FPGA-Research/FABulous/blob/FABulous2.0-development/FABulous/fabric_files/FABulous_project_template_verilog/user_design/top_wrapper.v) to instantiate the I/O BELs.

Is there a way to simply specify `input`/`output` in the top module and use a Physical Constraints file (.pcf) to map each `input`/`output` to a physical location?

I discovered that when a fabric is generated, FABulous generates a PCF under `.FABulous/template.pcf`. In addition, I found the `-iopad` option of the [`synth_fabulous`](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/synth_fabulous.html#cmd-synth_fabulous) script and it does seem to map from `input`/`output` to `IO_1_bidirectional_frame_config_pass`.

Is support for PCF in yosys/nextpnr in progress or is it perhaps already complete? I couldn't find anything about it in the documentation.

Two requests:

1. If the flow for PCF is already working, could this be used in the official example?
2. Can it be made possible to use a different User BEL during the `-iopad` call?

Thanks!

KelvinChung2000, 2025-01-23:
The PCF-related information is [here](https://github.com/YosysHQ/nextpnr/blob/master/docs/constraints.md). We are generating them but not supporting them yet.

mole99, 2025-01-24:
@KelvinChung2000 Thanks for the info!

What would it take to officially support PCF in FABulous? It would really make a difference in showing that FABulous is a proper FPGA flow to the end user.

A psotove side effect: The `run_FABulous_bitstream` command in the CLI currently implicitly specifies `top_wrapper` as top module. But with a proper PCF setup, the user design can actually be the top level, eliminating the need for this abstraction.

KelvinChung2000, 2025-01-24:
Take the ice40 as an [example](https://github.com/YosysHQ/nextpnr/blob/master/ice40/pcf.cc). It is mainly about parsing the file, it seems like this is primarily parsing, and nextpnr already has the API to add the information to the routing graph.

mole99, 2025-01-24:
Thanks, it's good to know what's missing!
Just FYI, I am not actively working on PCF support, I just wanted to understand and document the current state of it.

KelvinChung2000, 2026-02-10:
https://github.com/YosysHQ/nextpnr/pull/1637