FPGA-Research/FABulous #294: Separate file parsing from object creation

mole99, 2025-01-24:
Currently, FABulous is quite inflexible in the way it creates the `Fabric` and `Tile` objects:

`parseFabricCSV(fileName: str) -> Fabric` read in a CSV file and return a fabric object.

`parseTiles(fileName: Path) -> tuple[list[Tile], list[tuple[str, str]]]` read in a CSV file and return tile objects.

In my case, I want to create the fabric programmatically. I can now either write a CSV file just for FABulous to read it back in (which I currently do and want to get away from), or I can create the `Fabric` object myself.

In the latter case, I have to make sure the `Fabric` object is set up correctly, which is a lot of work and basically duplicates what is done in `parseFabricCSV`. Also, any changes to the way how FABulous creates the fabric will mean that this approach breaks.

---

Therefore, my suggestion is the following:

Instead of directly processing a file to create the object, first create a Python [dataclass](https://docs.python.org/3/library/dataclasses.html) that holds all of the necessary information. A fabric could then be created directly by passing that dataclass object to its constructor.

What I mean in pseudocode:

```Python
@dataclass
class FabricConfiguration:
    name: str
    num_columns: int
    num_rows: int
    ... # all config options

# Create the config object
my_fabric_config = parseFabricCSV(filename)
# or 
my_fabric_config = parseFabricYAML(filename)
# or
my_fabric_config = FabricConfiguration(name="my_fabric", num_columns=10, num_rows=10, ...)

# Create the fabric object
my_fabric = Fabric(my_fabric_config)
```

This has many advantages, such as:

1. An easy way to support new file formats
2. The config object can be created manually (my use case)
3. All configuration options that the fabric supports are in one place

The same approach could be used for Tiles. While I don't have a use case yet where I need to create tiles programmatically, the points from above also apply here.

Thanks!

KelvinChung2000, 2025-01-24:
I think the better approach might be to extend the FABulous API to allow setting those parameters because the Fabric itself is already a dataclass, and we should not need another layer of abstraction.

mole99, 2025-01-24:
You're right! I didn't realize that `Fabric` is already a dataclass.

What do you mean by "extend the FABulous API"? As mentioned above, I can already manually instantiated a `Fabric` object, but it is needlessly complicated to do so. Maybe the `Fabric` class can be cleaned up and the `commonWirePair` generation be moved into `__post_init__`?

mole99, 2025-01-24:
Yes, but that's exactly my point: It *should* be easy to set up the fabric programmatically. If the CSV file can create a new fabric object with all its information, then it should also be possible to do the same thing programmatically with the exact same information (without any extra work, I'm looking at you, `commonWirePair`).

I am not a huge fan of "creating a lot of helper functions that help build up the fabric".
It would actually suffice if was a bit easier to create the `Fabric` object. It should only need the information that is in the CSV file + the tile objects.

My not-so-great solution was to add another layer of abstraction that would make that easily possible.
But I agree, the better approach is to refactor the `Fabric` generation. Of course, this may take some time.