FPGA-Research/FABulous #296: Tiles with no config bits have no `FrameData` signals

mole99, 2025-01-24:
This may seem like a feature at first: Tiles that do not use any configuration bits have no `FrameData`/`FrameData_O` signals. But for the end user this is a actually a bug:

Tiles without any config bits and tiles with config bits can be mixed in a row. This requires all tiles to have `FrameData`/`FrameData_O` signals, even those without any config bits. Currently, this will result in a broken fabric.

IAmMarcelJung, 2025-01-24:
> Currently, this will result in a broken fabric

This sounds really concerning, can you explain what you exactly mean by "broken" here? Like broken not working or broken with weird side effects or parts of the fabric not working? Or something else? Either way, this shows that we really need to improve our testing infrastructure to catch such things (and for many other reasons ofc).

mole99, 2025-01-25:
@IAmMarcelJung it's "broken not working", as in the generated HDL code is not valid.

The reason for that is, if a tile has no config bits, then no `FrameData`/`FrameData_O` ports are generated. Thus, when the fabric is generated, the `FrameData` wires are connected to the non-existent `FrameData`/`FrameData_O` ports of those tiles.

See #297 for why this "feature" was added and why I think it's just more trouble than it's worth.

IAmMarcelJung, 2025-01-26:
I read your extensive descriptions in #297 and #161 (thanks!) and things got a bit clearer to me now (I'm honestly not too deep into how the fabric works). Did I get it right that this does ***not*** cause an issue if the first and last row are as it is currently in the demo fabric (since I could also not see the issue you described there) 

However you are ofc totally right that this should be fixed to work for all kinds of tiles and tile combinations, probably by the suggestions you made :)

mole99, 2025-01-26:
Yes, that's exactly what it means :)
I think this feature was implemented this way specifically for the demo fabric, but presents a problem for many other fabrics.

Sounds good, I've already made the necessary changes on my end. Now I just need to clean them up and generalize the handling of `NULL` tiles before I open a PR 👍

mole99, 2025-02-05:
Resolved by #306

PR #306 (2025-01-30): Improve FrameData and FrameStrobe handling
This PR fixes/improves how FrameData and FrameStrobe signals are connected within the fabric.

1. Tiles: Always generate FrameData ports
    - Fixes #296
    - This allows mixing tiles with no config bits and tiles with config bits in the same row
2. Remove special handling of first and last row
    - Fixes #297
    - This allows to have tiles with config bits in the first and last row (e.g. `N_IO`, `S_IO`)
3. Generalize handling of NULL tiles
    - Supersedes #161
    - A major improvement to how NULL tiles are currently handled

The general behavior [... cut]
closed 2025-02-05 by mole99