FPGA-Research/FABulous #302: Shell is entered instead of throwing an error when the script file is an existing directory

IAmMarcelJung, 2025-01-25:
**Describe the bug**
As mentioned in #301, the shell is entered if an existing directory is given instead of a `.tcl` file. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `FAB_ROOT` and create a new project or use an exisiting onr
2. Execute command with a wrong script argument (e.g. strip off the "l"):
 `FABulous <project_name> -fs=./<existing_directory>/`
3. The FABulous shell is entered

**Expected behavior**
I would expect the tool to throw an error instead of entering the shell.

**My Setup:**
 - OS: Ubuntu 24.04 LTS
 - Kernel version: 6.8.0-51-generic
 - Python version: 3.12.03
 - FABulous version: dev branch @ fd77c39e458ed59271cef18a83d7cde3ca483167

KelvinChung2000, 2025-01-27:
The file type check should not be required. In UNIX, file extensions are more like "for reference" than a requirement. 

I have recently started considering removing all the file extension checks based on the previous reason. Since you mentioned this, it might be a good time to work on this.

IAmMarcelJung, 2025-01-27:
I was referring to the extenstion, but rather to that if a an **existing directory** is given, FABulous should not be entering the shell (I corrected the text in step 2 of the reproduction description).

EverythingElseWasAlreadyTaken, 2025-08-13:
Fixed

closed 2025-08-13 by EverythingElseWasAlreadyTaken