FPGA-Research/FABulous #303: CLI: `run_simulation fst` does not create FST file

mole99, 2025-01-26:
The CLI doesn't pass the `-fst` option to vvp when `run_simulation fst ./user_design/sequential_16bit_en.bin` is called.

Without this option, vvp still creates a VCD file with the `.fst` file ending.

See: https://steveicarus.github.io/iverilog/usage/vvp_flags.html#vcd-fst-lxt-arguments

---

Quick question: While running the simulation with the CLI, I noticed that the output isn't correct:

```
fabric(I_top) = 0x000xxxx gold = 0x000xxxx, fabric(T_top) = 0xffffffe gold = 0xffffffe
```

When I run the emulation/simulation directly from the `Test/` directory, it works as expected:

```
fabric(I_top) = 0x48c018c gold = 0x48c018c, fabric(T_top) = 0xffffffe gold = 0xffffffe
```

Could someone check whether this is due to my environment or if you are seeing the same issue? This is on the `FABulous2.0-development` branch.

KelvinChung2000, 2025-01-27:
This could be something I broke during the cmd2 transition.

IAmMarcelJung, 2025-02-07:
Fixed in #290.

closed 2025-02-07 by IAmMarcelJung