FPGA-Research/FABulous #330: Make bit_gen an own project

EverythingElseWasAlreadyTaken, 2025-02-19:
The [bit_gen](https://github.com/FPGA-Research/FABulous/blob/FABulous2.0-development/FABulous/fabric_cad/bit_gen.py) program is a standalone program, that is mostly independent of the FABulous fabric generation, so it should be moved to an own repo, with an own documentation and an own development workflow. 

It should be also usable as a standalone program and include the [make_hex](https://github.com/FPGA-Research/FABulous/blob/FABulous2.0-development/FABulous/fabric_files/FABulous_project_template_vhdl/Test/makehex.py), to directly generate the hex files. 
This allows the user to run a full user design bitstream flow, without having to deal with FABulous itself, and also integrate it better in their own workflows. (Ofc. the user still needs bitstreamspec, pips and bel files etc. )

It should also be usable as a python dependency, to get a better integrated in the FABulous CLI. 

We should maybe rename bit_gen to something more meaningful and add it to pypi for better integration and better user experience.

KelvinChung2000, 2025-02-19:
I think we should move the bit_gen to become a part of the FABulous API rather than make it standalone. If there is a desire for bit_gen to be used as a standalone, write a tool that uses the API.

mole99, 2025-02-21:
I 💯 percent agree with @EverythingElseWasAlreadyTaken here. This is exactly what I wish for as a FABulous user.

So I can simply distribute the bundled fabric files (bitstream spec, pips, bels...) and point the user to upstream `yosys` + `nextpnr` + the *standalone* `bit_gen` program.

In fact, this is what I am already doing. I have a separate repository with just the user designs and I copied over the `bit_gen` script, which is not pretty. Already today I had the issue of not having the latest version of it in the repo (with support for the sync bit).

For the name of the PyPI package I would imagine something along the lines of `fabulous_utilities` as I can image it to include more functionality in the future such as: pack (already there), unpack, pll configuration etc.

@KelvinChung2000

> If there is a desire for bit_gen to be used as a standalone, write a tool that uses the API.

I have that desire, and I don't want to write a tool just to be able to generate a bitstream.

See any of the FPGAs supported by the open source tools e.g. iCE40 or ECP5: You install yosys, nextpnr and the corresponding bit gen program (`icepack` or `ecppack`) and that's it.

Very simple for the end user.

KelvinChung2000, 2025-02-21:
Of course, we package it out as well. Wrapping the API with argparse.

mole99, 2025-02-21:
As I mentioned before, I would prefer *not* to need FABulous and all its dependencies just to create a user bitstream.

However, as Jonas pointed out, `fabulous_utilities` (or what we want to call it) can be a dependency of FABulous and then you could use it from the API if you must.

KelvinChung2000, 2025-02-21:
That makes sense.