FPGA-Research/FABulous #346: docs: unclear FAB_ROOT command format

TaoBi22, 2025-03-25:
Currently the nextpnr compilation docs give examples like this:

`FAB_ROOT=<Project_directory> nextpnr-generic --uarch fabulous --json <JSON_file> -o fasm=<output_FASM_file>`

That `FAB_ROOT = x` at the start of the command is pretty confusing to me. Maybe there's some shell on which this works (?), or maybe it's just illustrative, but that's definitely not guaranteed, and people are probably expecting to be able to just copy and paste the commands there.

KelvinChung2000, 2025-03-25:
This is trying to set a temporary environment variable, which is then used by nextpnr. I believe this is a common shell feature since I see many compilation scripts doing something similar.

TaoBi22, 2025-03-25:
Ah I got the syntax slightly wrong when I tried it so I underestimated support. Cool, happy to be overruled if it's something done in other compilation scripts!

closed 2025-03-25 by TaoBi22