FPGA-Research/FABulous #349: genfasm command failing in VPR flow

muthu-iisc, 2025-04-05:
- genfasm command available in the VTR utilities is failing to generate a fasm file.
- Before this step I have successfully generated the place and route files for my design using the VPR tool.
- The fasm file is needed to generate the bit stream using BITMAN in the FABulous tool
- I have attached the screenshots of the above mentioned error with this query
- Could you please help me in resolving the issue?

Note:
Fabulous supports VPR flow according to the documentation.
genfasm is the utility used for generating the fasm file which will be required for bitstream generation.
I followed the steps mentioned in the VPR compilation page in the fabulous website.

[image]
[image]

Thanks,
Muthiah

KelvinChung2000, 2025-04-05:
Using the VPR `genfasm` will not work. You must use the `bit_gen.py` we provide to generate the bitstream.

KelvinChung2000, 2025-04-05:
I suggest not using the VPR flow but using the nextpnr flow. We have more support on the nextpnr side.