FPGA-Research/FABulous #356: Auto Generate User Design Top Wrapper Depending on Fabric Size

EverythingElseWasAlreadyTaken, 2025-05-08:


yannickreiss, 2025-05-17:
I have something like that for my design and tiles, is it already wip? Else I could provide my approach as a base.

It currently needs providing I/O information in the fabric.csv tile definition, but if it could be usable in another project, I would adjust it to rely on the current fabric.csv syntax.

yannickreiss, 2025-05-17:
https://gitlab.cs.hs-rm.de/yreis001/riscvar/-/wikis/Tools/FABulous/Create_Constraints

This is a reference to what I built. Like I said, it won't be very useful outside my design, but could be adjusted easily, I think.

EverythingElseWasAlreadyTaken, 2025-05-17:
Hey,

thank you very much for sharing your script!
I'll take a look at it. :) 

We are already working on a few things to make the whole IO handling in general easier :)