FPGA-Research/FABulous #357: Problems with changing the demo (number of W_IO)

yannickreiss, 2025-05-11:
I am trying to adjust the demo project to an eFPGA with 74 to 106 IOs.

## The problem

When running the fabulous_tb testbench, I_top and T_top are marked unitialized, as shown below:

```
fabric(I_top) = 0xxxxxxxxxxxxxxxxxxxxxxxxxxxx gold = 0x000000003fff400000000000000, fabric(T_top) = 0x000ffffffffxxxxXxxxxxxxxxxx gold = 0x0000000000000000000ffffffff
fabric(I_top) = 0xxxxxxxxxxxxxxxxxxxxxxxxxxxx gold = 0x000000003fff800000000000000, fabric(T_top) = 0x000ffffffffxxxxXxxxxxxxxxxx gold = 0x0000000000000000000ffffffff
fabric(I_top) = 0xxxxxxxxxxxxxxxxxxxxxxxxxxxx gold = 0x000000003fffc00000000000000, fabric(T_top) = 0x000ffffffffxxxxXxxxxxxxxxxx gold = 0x0000000000000000000ffffffff
```

## Attempts to solving

- Changed wire/reg sizes in fabulous.tb
- Added new W_IO tiles to top_wrapper (Test + user_design)
- Changed the test designs
- Changed max bitstream length to ~32k
- Change number of rows with NULL instead of W_IO

The problem comes when changing the amount of W_IO tiles.

Thanks in advance for any help

## Fabric

```csv
FabricBegin,,,,,,,,,,,,,,,,,,
NULL,N_term_single,N_term_single,NULL,,,,,,,,#,,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,180,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,,,,,,,
W_IO,LUT4AB,LUT4AB,NULL,,,,,#,,,,,,,
NULL,S_term_single,S_term_single,NULL,,,,,,,,#,2700,,,,,,
FabricEnd,,,,,,,,,,,,,,,,,,
```

EverythingElseWasAlreadyTaken, 2025-05-11:
Our current configuration logic limits us in terms of fabric size. 
We are already planing a new configuration logic, but this will take some more time. 

Our current limits for the fabric limits are: 
-  At most 32 columns
-   At most 32 rows
-  At most 20 frames per tile
-   At most 26 bels per tile

Here you can find some more information about these limits: https://github.com/FPGA-Research/FABulous/discussions/325

As an alternative, you could try to add IOs on the remaining sides of the fabric. 
If you give us more information about what you want to achieve, we can help you find a solution.

yannickreiss, 2025-05-13:
Reducing the rows partially fixed the problem.
Instead of uninitialized I_top values, they are now either '0' or 'x'.

Regarding the limits: Maybe they could be added to the Fabric Definition chapter of the documentation, or is there a reason against adding it?

Regarding what I want to achieve:

I want to integrate an eFPGA into a RISC-V processor, giving it enough I/Os for 2x32 bit registers input, 1x32 bit output and 10 additional inputs for a custom decoder.

Additionally, for creating the flow I use the following script:

```tcl
load_fabric
run_FABulous_fabric
gen_top_wrapper
run_FABulous_bitstream npnr ./user_design/ciu.v
exit

```

EverythingElseWasAlreadyTaken, 2025-05-13:
> Reducing the rows partially fixed the problem. Instead of uninitialized I_top values, they are now either '0' or 'x'.
Sounds like an issue with the user design top wrapper or your user design.
If you provide your FABulous project, I can take a look at it and see if I can find something. 

> Regarding the limits: Maybe they could be added to the Fabric Definition chapter of the documentation, or is there a reason against adding it?
We are currently working on a new Release, it will be added with it. 

 > Regarding what I want to achieve:
> 
> I want to integrate an eFPGA into a RISC-V processor, giving it enough I/Os for 2x32 bit registers input, 1x32 bit output and 10 additional inputs for a custom decoder.

It seems that you don't need the tristate drivers that are implemented in the W_IO for this. 
You can take a look how the RAM_IO is implemented, these could be a better boilerplate for your IOs. 

If you still require more IO, you can also take a look how we utilized the N and S Term tiles to implement the USER_IOs here:
https://github.com/Asma-Mohsin/FPGA_IGNITE_2024/tree/main/FABulous/CXBex_fabric/Tile/S_term_single
But this should be tested carefully and will increase the risk for routing congestions in the user design.

You can also take a look at the Greyhound SoC, which implements something similar to what you have in mind:
https://github.com/mole99/greyhound-ihp

ninachloekassi, 2025-05-14:
Thank you very much for all your help.
As the original issue was resolved with the reduced amount of rows, I'll close it now.
I am continuing with custom cells.

closed 2025-05-14 by ninachloekassi