FPGA-Research/FABulous #362: Decide on a versioning system

KelvinChung2000, 2025-05-22:
#270, #322, and #328 have started the discussion, but we really should finalise this.

mole99, 2025-05-23:
Thanks for getting the ball rolling.

I'll summarize what was discussed so far:

[Semantic versioning](https://semver.org/): MAJOR.MINOR.PATCH

- `main` - stable branch (rename from current `master`)
- `dev` -  development branch (rename from current `FABulous2.0-development`)

Features are developed in feature branches. These are continously merged into `dev`. Once enough features have accumulated, `dev` is merged into `main` (after thorough testing).

Either manually publish PyPI releases for `main` or set up a CI to do so whenever the version in `main` changes.

Add a changelog to the repository. The changelog is expanded with every feature and bugfix.

EverythingElseWasAlreadyTaken, 2025-05-23:
> Thanks for getting the ball rolling.
>
> I'll summarize what was discussed so far:
>
> [Semantic versioning](https://semver.org/): MAJOR.MINOR.PATCH
>
>     * `main` - stable branch (rename from current [...]

Sounds like a good start.
Could be, that we need a major and minor dev branch at some point, but let's see how far we get.

I'd like to add a few things to this: 

- The master branch should be always stable and tested, but not ever PR on the master needs to be a release.
- The dev branch can be unstable.
- Releases are done by setting a tag to a release commit of the main branch.
    - This should automatically trigger a release build and a deployment to PyPi 
    - Release deployment to PyPI needs to be confirmed by a different maintainer. 
    -  [Optional] Every PR merged to dev branch, can also automatically trigger a deployment to Test-PyPi, and runs a test workflow afterward, to check if everything is working.

I've already prepared the deployment workflows.

- We should go for a <major>.<minor>.<hotfix> versioning scheme, eg. FABulous-1.0.1
    - Major releases can involve breaking changes and bring substantial changes in the codebase or workflow.
    - Minor releases can bring new features and not urgent fixes, but noting breaks, backward compatibility is required.
    - Hotfix releases are just for critical or urgent bug fixes. 
    - I also would like to keep the dirty/post versioning tags from the setuptools git versioning for development / nightly builds:
        - https://setuptools-git-versioning.readthedocs.io/en/stable/schemas/tag/dirty_version.html

- Changelogs should be automatically generated based on all the headlines of all commit messages, involved in a release
    - This would require that we use the semantic versioning scheme and require it for all commits.
    - Squash PRs should be only used for smaller changes. 
    - Bigger changes should be split in multiple, self containing commits, with a meaningful commit message

EverythingElseWasAlreadyTaken, 2025-07-21:
I've updated the existing tag of FABulous 1.0.0 to follow semantic versioning and updated the pyproject.toml for better support of semantic versioning. #396

I've also prepared a new release for the current master.

Afterward, we can plan the 1.5.0 release of the current dev branch.

closed 2025-07-23 by EverythingElseWasAlreadyTaken