FPGA-Research/FABulous #363: Unify BelMap feature for VHDL and Verilog

EverythingElseWasAlreadyTaken, 2025-05-23:
The BelMap feature is quite different for VHDL and Verilog bels:

```
Verilog:
 (*FABulous, BelMap, INIT=0, INIT_1=1, INIT_2=2*)
VHDL:
 -- (* FABulous, BelMap, INIT=0, INIT[1]=1, INIT[2]=2,*)
```

The Verilog implementation only supports vectorization for the `INIT` keyword and uses a `_` as separator to set the index.
The VHDL implementation seems to support vectorization for any keyword and uses the usual vector notation.

This should be unified to use the same representation in both languages.
We should support vectors for any keyword.
I also like the vector notation more, since it expresses that the parameter is actually vectorized.
Also, the first value  (index 0) should be already defined in vector notation, it feels kinda odd to have the first index without it. 

We also need to update all bels accordingly.

KelvinChung2000, 2025-05-23:
The problem is that Verilog does not support regular vector syntax for attributes. Will need to think about how to make the unification possible.

EverythingElseWasAlreadyTaken, 2025-05-25:
> The problem is that Verilog does not support regular vector syntax for attributes. Will need to think about how to make the unification possible.

Ok, but then we should also go for the underscore naming scheme for VHDL.
A common one, that is a bit odd, is better than having two different naming schemes.
We should also add backwards compatibility or at least meaningful errors/warnings for legacy projects.