FPGA-Research/FABulous #374: I/O Tile order in top wrapper.

yannickreiss, 2025-06-06:
I know, there is a script for top wrapper generation in FABulous 2.0, but until that is on main and documented, I focus on my own.

# Situation

I created a mirror tile to W_IO called E_IO exactly mirrored.

## Problem

The order of the tiles seem to cause problems.

Right now I map them in the following order:

- West
- East
- Move down west
- West
- East
- ...

## Example of the problem

I got 105 I/O ports. 
The oeb and gold values are:

- 0x03ffffffffffffffffff0000000
- 0x3ffffffffffffffffff00000000

## Question

What is the right order of I/O tiles?
If it is documented, where?

yannickreiss, 2025-06-10:
That is the problem. I generate the top_wrapper myself until the builtin generator is documented. Using my custom E_IO tile, the problem with that is, that the test design and top_wrapper are different. I started to switch IO-Tiles in the top_wrapper, but figured if I had the rule for that, it would spare me much time.

IAmMarcelJung, 2025-06-10:
Hi, what is the acutal issue you're encountering? You mentioned that the test design and the top wrapper are different. Currently both need to be adjusted if you change the default fabric. As Kelvin mentioned you are free to choose how you map the IOs in your user design to the tiles, so there is no blueprint for this. Could you maybe share your code so we can see what you are trying to achieve and where the problem arises?

IAmMarcelJung, 2025-06-10:
Thanks for sharing your top wrapper. I think it would make sense to assign the IO's block by block, so e.g. `X0Y1.B` -> `io[0]` to`X0Y27.A` -> `io[53]` and `X8Y1.B` -> `io[54]` to `X8Y27.A` -> `io[107]` (just an example, adjust it as it makes sense for your use case). I'm sorry, but I still don't 100% understand the problem. Are you getting any errors in a specific step?

> some ports seem to be in a different order

To what order are you referring to? The order in the top wrapper vs the order in the user design vs the order in the test bench? If yes, then this has to be adjusted by you so the assignments match.

>  some io ports are switched

How do you expect them to be assigned? Do you have an example of such switched ports? 

It could also be helpful to additionally share your `fabric.csv`, the user design and the testbench. I looked for these files in [this repo](https://gitlab.cs.hs-rm.de/yreis001/riscvar) but I could not find the files that match the top_wrapper you just shared.

Also is there a specific reason why your io wires are 212 bits wide?

Unrelated to your current issue: Are you sure you want to use this many IO's? Usually you are quite limited in terms of IO Pads when taping out a chip, so this could be quite a challenge. As it was stated in [this answer](https://github.com/FPGA-Research/FABulous/issues/357#issuecomment-2876295337) in #357, you could consider alternatives to using the Tri-state IO primitives. But if this is all intended, we can of course try to help you make it work as you intend it!

EverythingElseWasAlreadyTaken, 2025-06-11:
Could you please provide your whole demo project, that we try out the simulation and play around with it? 
Please add a quick description how to run it, if you changed something from the default flow.

yannickreiss, 2025-06-11:
```bash
git clone https://gitlab.cs.hs-rm.de/yreis001/riscvar.git
cd riscvar
git checkout create_design_from_E_IO
make setup
make simulatefpga
```

`make setup` is used to install FABulous

`make simulatefpga` is used to copy the directory eFPGA into FABulous, run the flow and then the simulations.

Thank you very much for the support.

IAmMarcelJung, 2025-06-12:
Thanks for providing your project information. One thing you can try is to use separate vectors for the `W_IO` and the `E_IO` tiles. Then assign these vectors similar to how it's done in the demo project, by assigning the LSB to the BEL with the highest `Y` value (`X0Y27.B` for `W_IO` and `X8Y26.B` for `E_IO` in your case) and the MSB to the BEL with the lowest `Y` value (`X0Y1.A`/`X8Y1.A1`). While the assignment can in general be chosen freely, this might be the easiest way to keep it consitent between the top wrapper, the user design and the test bench.

yannickreiss, 2025-06-20:
Solved problem by moving all ports to west side of the fabric.
When I got time, I will try to actually identify and solve the problem.

Thanks again very much for the help.