FPGA-Research/FABulous #435: FlipFlopChain mode of ConfigBitMode generates an incorrect netlist

AndreaMazzan, 2025-08-19:
**Describe the bug**
I am running the demo project using the Docker environment for FABulous. If I change `ConfigBitMode` from `frame_based` to `FlipFlopChain` in the file `demo/fabric.csv`, the generated Verilog netlist contains syntax errors, undeclared variables, and invalid module errors. Iverilog picks up these errors during the simulation of the simple counter example present in the `Test/test_design/` folder

**To Reproduce**
Steps to reproduce the behaviour:
1. Open Docker env
2. Execute commands `python3 FABulous.py -c demo` and `python3 FABulous.py demo`
3. In the FABulous shell, run the commands `load_fabric` `run_FABulous_fabric` and `exit`
4. Navigate to the test folder `demo/Test` and run `./build_test_design.sh` and `./run_simulation.sh`
5. See multiple errors affecting every tile

**A Bit of Debugging**
I went and checked the errors that Iverilog is throwing.
If we take for example the LUT4AB tile, one error is
```
tmp/LUT4AB.v:1672: syntax error
tmp/LUT4AB.v:1672: error: Syntax error in left side of continuous assignment.
tmp/LUT4AB.v:1673: syntax error
tmp/LUT4AB.v:1673: error: Syntax error in left side of continuous assignment.
```
These lines of code correspond to these statements
```
assign conf_data(conf_data'low) = CONFin; //conf_data'low=0 and CONFin is from tile entity
assign conf_data(conf_data'high) = CONFout; //CONFout is from tile entity
```
There are two issues here, the syntax is a mix of verilog and VHDL, and the variable conf_data is never delcared as a wire or a reg.

interestingly, I noticed that in `LUT4AB.v`, the LUTs are instantiated as

```
LUT4c_frame_config_dffesr Inst_LA_LUT4c_frame_config_dffesr (
    .I0(LA_I0),
    .I1(LA_I1),
    .I2(LA_I2),
    .I3(LA_I3),
    .Ci(LA_Ci),
    .SR(LA_SR),
    .EN(LA_EN),
    .O(LA_O),
    .Co(LA_Co),
    .UserCLK(UserCLK),
    .MODE(Mode),
    .CONFin(conf_data[0]),
    .CONFout(conf_data[1]),
    .CLK(CLK)
);
```
However, the module `LUT4c_frame_config_dffesr ` found in `LUT4c_frame_config_dffesr .v` does not have the last 4 ports declared above

```
module LUT4c_frame_config_dffesr (I0, I1, I2, I3, O, Ci, Co, SR, EN, UserCLK, ConfigBits);
```

So some wires are not properly connected in the overall netlist.

The other error that Iverilog finds is
```
tmp/LUT4AB_switch_matrix.v:996: syntax error
tmp/LUT4AB_switch_matrix.v:996: error: Syntax error in continuous assignment
tmp/LUT4AB_switch_matrix.v:996: syntax error
tmp/LUT4AB_switch_matrix.v:998: error: Invalid module item.
tmp/LUT4AB_switch_matrix.v:998: syntax error
tmp/LUT4AB_switch_matrix.v:998: error: Invalid module instantiation
tmp/LUT4AB_switch_matrix.v:998: error: Invalid module instantiation
tmp/LUT4AB_switch_matrix.v:1009: syntax error
tmp/LUT4AB_switch_matrix.v:1010: error: Invalid module item.
```

These errors refer to this block of code in the switch matrix

```
//The configuration bits (if any) are just a long shift register
 //This shift register is padded to an even number of flops/latches

    genvar k;
    assign ConfigBitsInput = {ConfigBits[462-1-1:0], CONFin;}
    // for k in 0 to Conf/2 generate
    for (k=0; k < 461; k = k + 1) begin: L
        LHQD1 inst_LHQD1a(
            .D(ConfigBitsInput[k*2]),
            .E(CLK),
            .Q(ConfigBits[k*2])
        );
        LHQD1 inst_LHQD1b(
            .D(ConfigBitsInput[(k*2)+1]),
            .E(MODE),
            .Q(ConfigBits[(k*2)+1])
        );
    end
    assign CONFout = ConfigBits[462-1];
```

Unfortunately, I don't fully understand what this block is supposed to be doing. First, `ConfigBitsInput` and `ConfigBits` are not defined as wires or regs (or IOs). Secondly, the iteration runs up to 461, meaning that at the very least, there is going to be an `index out of bounds` error when `ConfigBitsInput[2*k]` is called (Unless `ConfigBitInput` is meant to be double the size of `ConfigBits`). Lastly, this doesn't really look like a shift register to me, as the latches are not interconnected, and I would have expected FlipFlops.

Another thing that I found interesting is the memory of the tile `LUT4AB_ConfigMem`, which looks the same as the memory in the `frame_based` configuration, so it seems there really isn't a FlipFlop chain that stores the bitfile.

**Additional Comments**
I originally opened a discussion about this issue, believing that I was using the wrong BELs, but apparently, changing the value of `ConfigBitMode` should be enough for the software to generate a functional netlist.
I tried downloading the 1.0 release version but I encountered similar issues.

Thanks in advance!

KelvinChung2000, 2025-08-19:
I'm sorry, but I haven't been able to get around to this issue yet. I cannot promise the fix will be anytime soon, as it might be non-trivial. Is there anything specifically you'd like to achieve? We can help you work out a solution that works for you in the meantime.

AndreaMazzan, 2025-08-22:
Hi all,
Thank you for your quick response. I was under the impression that a long shift register would be more area-efficient, but I understand from Dirk's comments that this might not be the case at all.
I am very thankful for the support and the suggestions related to radiation hardening techniques.
DICE cells would be amazing in principle and we are aiming to expand our project to integrate custom DICE SRAMs (or DICE FF) into the design, but we thought we could start from TMR to validate the approach first (this is our first time using FABulous).
I would be delighted to have a call, I will reach out to organise that.
Thank you,
Andrea